emoncms/emoncms #898: New First User Creation Error

kumarmohan7, 2018-06-26:
Hi,

I create New 1st user creation i got error "[object Object]",

How to solve this error,
Please help to solve this error,

I am now using Shared Host

I attached settings.php & Error Screenshot.
[settings.php](https://github.com/emoncms/emoncms/files/2138492/settings.txt)
[image]

TrystanLea, 2018-06-26:
Hello @kumarmohan7 , could you check your Apache logs? /var/log/apache2/error.log?

TrystanLea, 2018-07-10:
closing this issue as it is associated with your latest issue which looks like the underlying cause https://github.com/emoncms/emoncms/issues/912

closed 2018-07-10 by TrystanLea