emoncms/emoncms #903: Can MQTT connection happen outside of publish to mqtt process?

TrystanLea, 2018-07-01:
Can MQTT connection happen outside of publish to mqtt process? see:

See: https://community.openenergymonitor.org/t/mqtt-log-entries/7784/4

TrystanLea, 2018-07-01:
Perhaps the process model can reuse the existing mqtt connection started by the phpmqtt_input script or if not available e.g when being used as part of the HTTP route, it could start a single publishing client if connect was called in the _construct rather than the publish_to_mqtt method itself

Paul-Reed, 2018-07-04:
I've moved a couple of my comments here, to try and keep the conversation in one place!!
> In https://github.com/emoncms/emoncms/pull/907 @emrysr wrote;  - the publish and subscribe calls are now done only done if connected.

But they are still accumulating in the queue, and when connected, the whole lot of backdated data is released to the broker, which could be hours worth.

In https://github.com/emoncms/emoncms/pull/908 the clientid has been reverted, but Nick O'Leary (node-red author) got around this for his [pubsubclient mqtt library](https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_esp8266/mqtt_esp8266.ino) by prefixing a random string with the client ID name, so it can easily be recognized.
ie `EMONCMSclient-3b77e8`

The C++ code is below;

```
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str())) {
      Serial.println("connected");
```

pb66, 2018-07-05:
from #907 @TrystanLea asked

> Thanks @pb66 should we revert the added client id for now, 

IMO it might be marginally better to have a correctly working QoS 0 implementation than a partial QoS 2 implementation resulting in something similar but not entirely like a QoS 0 set up.

Having a client id is the right thing to do, having a QoS of 0 when there is no timestamp provided with the input data is the right combination regardless of client id.

> I would prefer not to add a different issue to these particular changes relating to publish_to_mqtt?

That's fine to want to have sepearte PR's but we do need to discuss the MQTT as a whole, I have created a mqtt branch so that we can perhaps test several things and try to resolve several things before merging quick hotfixes to the master repo and reducing our options.

in #905 @TrystanLea said

>  Can anyone think of a better way to solve the buffering issue?

In the  mqtt branch I have expanded on my previous suggestion from here (https://github.com/emoncms/emoncms/pull/905/commits/9f0dc1199860053ff52db41f8b93e8acdbd7a365) to pop single items and loop faster.

I have also made it so that only data less than 2mins old will get published when published using QoS0 "status" type single value payloads.

There is some basic (not intended for release) code for higher QoS levels to demo where it could sit and how it might work.

there is also a TODO marker for a log entry to be added to log the queue length, but that will need to be throttled so it doesn't log every loop iteration, and the log level should reflect the queue length too.

also from #905 @Paul-Reed said

> before adding mqtt data to the buffer, could a check be made to see if connected to the broker? and only cache data if connected, otherwise dump.

IMO right idea but wrong place. I think all queue management and mqtt should be in the phpmqtt?!*@{} script. I have included a check to see if the data is less than 2mins old (IF using QoS0) and the data gets dumped otherwise. This is a tricky solution that's less than ideal because you could still get a flurry of emonPi values that are getting stale, but if we go less than 2mins we might not be able to reliably publish emonTH data. It is much better than getting hours, days or weeks of values published though.

@borpin said

> You could also mitigate the burst issue by limiting the number of publish requests done per cycle. Into this would also be the QoS as the handshaking will take longer I suspect. Any limit on the publish rate should be clear.

I agreed the limit should be clear, but it maybe better to make it a user setting as some users might be collecting HH data (for example) whilst others need a more exact "current" status for controlling stuff and want to reduce that to 10secs etc .

My suggested code only does one item per iteration, this avoids "bursts" all together unless the loop is interupted by a subscription or a connection issue. spinning rapidly and only popping a single value means the loop is always responsive and the other checks in the loop will spread the publishes out by a few millis even is there is nothing else to do at the time, this gives the broker a breather too.

> Rather than a queue it just needs to be a 'latest' data value for now until the timestamp option is available.

That will not work either as the check can be done many times whilst emoncms is processing a backlog, eg instead of the original thousands of published values you might get one from every couple of hours as that was the "latest" in the queue when checked but only part of the backlog had been processed at that time and so on, it must be based on the timestamp to know if it's "current" or not.

The code in the mqtt branch is not tested in any way, I just did it freehand in notepad, so expect errors, it is more to show you what I'm thinking than a finished piece.

I have also commented out a large chunk of what appears to be mqtt code still in the process file, I do not understand why there is any need for any mqtt code in the process. I have just commented it out to see if it made any difference or if it is (or could be made) redundant.

>In #908 the clientid has been reverted, but Nick O'Leary (node-red author) got around this for his pubsubclient mqtt library by prefixing a random string with the client ID name, so it can easily be recognized.
ie EMONCMSclient-3b77e8

The clientid code was only reverted because it will cause issues unless the input (subscribe) QoS level is changed to QoS0 at the same time (or the input payload is altered to include a timestamp). It was previously decided to put a client id in the settings.php. There is no need for emoncms to randomise it's clientid and doing so would mean that restarting apache2/emoncms could result in a different client id which would lose all the data the broker has buffered whilst restarting apache2/emoncms.

see https://github.com/emoncms/emoncms/tree/mqtt

pb66, 2018-07-05:
Just added another commit to the mqtt branch to add these variables to $mqtt_server in settings.php
```
                          'clientid' => '',
                          'sub_qos'  => 2,
                          'pub_qos'  => 0,
                          'qos0_limit'=> 120
```
The empty clientid string and QoS2 should give us the same setup as was in place before we started these latest changes to the publish to mqtt.

The "pub_qos" of 0 and 120secs limit to "status" data carries forward the settings I included in the initial commit to the mqtt branch.

Breaking these out to the settings will allow us to try different setups, they do not necisarily need to stay, although I think they should. I also pondered on the "keepalive" setting but have not yet put that in.

 Looking at the code, it seems the clientid is still in
https://github.com/emoncms/emoncms/blob/163b377946037de1000b4e4acb4ad34d6b58cffa/scripts/phpmqtt_input.php#L105

Looking at the PR, it seems it was the (redundant?) publish connection in the process file that was reversed, not the subscribe connection in the phpmqtt file (see https://github.com/emoncms/emoncms/pull/908/files)

I have attempted to add a ternary if/else using the clientid _and_ QoS as a test, the intention is that if the clientid is empty or the QoS is 0 the cleansession is set to true and if a clientid is used with a Qos of 1 or 2, it is set to false to make full use of the brokers buffering.

I also corrected the logic to the published "if qos" as it was 180° out in the initial commit.

THIS IS STILL UNTESTED

pb66, 2018-07-05:
Also added some logging for publishing and also for queue length every 60s if a queue exists, the log level is "info" unless the queue is over 1000 then it is a warning.

STILL UNTESTED

pb66, 2018-07-05:
here's a link to the diff of all these changes compared to the master branch

https://github.com/emoncms/emoncms/compare/mqtt

Paul-Reed, 2018-07-05:
I realise that this is a POC, but is the mqtt branch working at your end @pb66 ?
Just tried it here and no messages are being published.
I have updated settings.php.
```
1530813584: mosquitto version 1.4.10 (build date Fri, 22 Dec 2017 08:19:25 +0000) starting
1530813584: Using default config.
1530813584: Opening ipv4 listen socket on port 1883.
1530813584: Opening ipv6 listen socket on port 1883.
1530813588: New connection from 192.168.1.8 on port 1883.
1530813588: New client connected from 192.168.1.8 as mqtt_7b361656.82ea78 (c1, k15, u'emoncms').
1530813588: Sending CONNACK to mqtt_7b361656.82ea78 (0, 0)
1530813588: Received SUBSCRIBE from mqtt_7b361656.82ea78
1530813588:     power (QoS 0)
1530813588: mqtt_7b361656.82ea78 0 power
1530813588:     solar (QoS 0)
1530813588: mqtt_7b361656.82ea78 0 solar
1530813588:     diverted (QoS 0)
1530813588: mqtt_7b361656.82ea78 0 diverted
1530813588:     test (QoS 0)
1530813588: mqtt_7b361656.82ea78 0 test
1530813588:     greenhouse/temp (QoS 1)
1530813588: mqtt_7b361656.82ea78 1 greenhouse/temp
1530813588:     greenhouse/control (QoS 1)
1530813588: mqtt_7b361656.82ea78 1 greenhouse/control
1530813588:     greenhouse/status (QoS 1)
1530813588: mqtt_7b361656.82ea78 1 greenhouse/status
1530813588:     greenhouse/rssi (QoS 1)
1530813588: mqtt_7b361656.82ea78 1 greenhouse/rssi
1530813588:     greenhouse/moist (QoS 1)
1530813588: mqtt_7b361656.82ea78 1 greenhouse/moist
1530813588:     emoncms/tempc (QoS 0)
1530813588: mqtt_7b361656.82ea78 0 emoncms/tempc
1530813588:     emon/# (QoS 0)
1530813588: mqtt_7b361656.82ea78 0 emon/#
1530813588: Sending SUBACK to mqtt_7b361656.82ea78
1530813588: New connection from 192.168.1.11 on port 1883.
1530813588: New client connected from 192.168.1.11 as ESP8266Client-ea07 (c1, k15, u'esp').
1530813588: Sending CONNACK to ESP8266Client-ea07 (0, 0)
1530813588: Received PUBLISH from ESP8266Client-ea07 (d0, q0, r0, m0, 'greenhouse', ... (11 bytes))
1530813588: Received SUBSCRIBE from ESP8266Client-ea07
1530813588:     greenhouse/control (QoS 0)
1530813588: ESP8266Client-ea07 0 greenhouse/control
1530813588: Sending SUBACK to ESP8266Client-ea07
1530813603: Received PINGREQ from mqtt_7b361656.82ea78
1530813603: Sending PINGRESP to mqtt_7b361656.82ea78
1530813603: Received PINGREQ from ESP8266Client-ea07
1530813603: Sending PINGRESP to ESP8266Client-ea07
1530813618: Received PINGREQ from mqtt_7b361656.82ea78
1530813618: Sending PINGRESP to mqtt_7b361656.82ea78
1530813618: Received PINGREQ from ESP8266Client-ea07
1530813618: Sending PINGRESP to ESP8266Client-ea07
1530813633: Received PINGREQ from mqtt_7b361656.82ea78
1530813633: Sending PINGRESP to mqtt_7b361656.82ea78
1530813633: Received PINGREQ from ESP8266Client-ea07
1530813633: Sending PINGRESP to ESP8266Client-ea07
1530813648: Received PINGREQ from mqtt_7b361656.82ea78
1530813648: Sending PINGRESP to mqtt_7b361656.82ea78
```

pb66, 2018-07-05:
Unfortunately I do not have anything I can test this with right now, so I haven't even tried it.

I will have another look through to see if there's something obvious. Perhaps you could try uncommenting the mqtt block at the top of the Modules/process/process_processlist.php file

https://github.com/emoncms/emoncms/blob/mqtt/Modules/process/process_processlist.php#L47-L66

and add "$mqtt_server" back into line 47.

If that works it means the process is still using some of that code and the changes I've made to the setting won't be working. I haven't tried to figure out exactly how it is/was working, I just commented out that code as I assumed it was now redundant.

Thanks for testing anyways :-)

pb66, 2018-07-05:
I have just added another commit that might make a difference.

I have changed the procees to (hopefully) use the client instance created in phpmqtt_input.php instead of the one I had commented out. But I'm not confident of "scoping" in PHP so I do not know if it will work first go (unlikely I guess )

Paul-Reed, 2018-07-05:
Yes, your last commit made the difference Paul, I have it working here.
Not sure if pub_limit is working though.

I've just stopped mosquitto for 8 minutes.
Emoncms publishes 3 synchronous messages every 5 seconds.
Upon restarting mosquitto, 285 messages were in the queue and processed by the broker.
So that's 285/3 = 95
95 messages at 5 secs apart = 7 minutes 55 sec

pb66, 2018-07-05:
doh! That should be "qos0_limit" not "pub_limit", It's right in default.settings.php but I see there's a typo in default.emonpi.settings.php, that must be where you got the setting from?

pb66, 2018-07-05:
Odd, so do you definitely have "qos0_limit" in settings.php? The only place I could see "pub_limit" was in default.emonpi.settings.php, where are you reading "pub_limit" from?

I've added a temporary log line to print out the 3 values used to limit the publishing,

the datapoint timestamp, the qos0_limit setting and the current time if you wouldn't mind running that and posting a line from the log that should be something like

Timestamp is 1234567890 and qos0_limit is 120 and time() is 1234567890

the values should only get published when the timestamp is greater than the current time minus the limit.

pb66, 2018-07-05:
Can you show us a bit of log?

Paul-Reed, 2018-07-05:
I've just re-run it for 4 minutes.
```
1530826284: mosquitto version 1.4.10 (build date Fri, 22 Dec 2017 08:19:25 +0000) starting
1530826284: Using default config.
1530826284: Opening ipv4 listen socket on port 1883.
1530826284: Opening ipv6 listen socket on port 1883.
1530826287: New connection from 192.168.1.11 on port 1883.
1530826287: New client connected from 192.168.1.11 as ESP8266Client-8f66 (c1, k15, u'esp').
1530826287: Sending CONNACK to ESP8266Client-8f66 (0, 0)
1530826287: Received PUBLISH from ESP8266Client-8f66 (d0, q0, r0, m0, 'greenhouse', ... (11 bytes))
1530826287: Received SUBSCRIBE from ESP8266Client-8f66
1530826287:     greenhouse/control (QoS 0)
1530826287: ESP8266Client-8f66 0 greenhouse/control
1530826287: Sending SUBACK to ESP8266Client-8f66
1530826289: New connection from ::1 on port 1883.
1530826289: New client connected from ::1 as emoncms (c1, k5, u'emoncms').
1530826289: Sending CONNACK to emoncms (0, 0)
1530826289: Received SUBSCRIBE from emoncms
1530826289:     emon/# (QoS 2)
1530826289: emoncms 2 emon/#
1530826289: Sending SUBACK to emoncms
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
[... 87 lines cut ...]
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'power', ... (3 bytes))
[... cut ...]
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'solar', ... (1 bytes))
1530826289: Received PUBLISH from emoncms (d0, q0, r0, m0, 'diverted', ... (1 bytes))
1530826292: New connection from 192.168.1.8 on port 1883.
1530826292: New client connected from 192.168.1.8 as mqtt_702d7e62.90527 (c1, k15, u'emoncms').
1530826292: Sending CONNACK to mqtt_702d7e62.90527 (0, 0)
1530826292: Received SUBSCRIBE from mqtt_702d7e62.90527
1530826292:     power (QoS 0)
1530826292: mqtt_702d7e62.90527 0 power
1530826292:     solar (QoS 0)
```

pb66, 2018-07-05:
Sorry, I meant emoncms.log, emoncms cannot log to mosquitto.log

pb66, 2018-07-05:
Just to confirm, your data is coming into emoncms via http correct? it has a timestamp? It has just  occurred to me this limit will be of no use with mqtt inputs as when the broker cannot deliver it will buffer the data and then a flood of non-timestamped inputs will all end up with a timestamp of now, so any publish to mqtt derived from those inputs will have a "now" timestamp too. Luckily in most instances users will not generally  have mqtt inputs publishing to mqtt, so the unwritten rule of thumb here will have to be mqtt inputs OR publish to mqtt but not both until the inputs get qos0 and/or timestamps.

To my knowledge, all emoncms,log timestamps are always UTC (ie GMT) and although the log timestamps are independent of the timestamps used by the inputs and feeds etc, they too should hopefully all be in UTC too. (TZ times should only really get used when displaying data or daily totals).

It was because of potential time issues eg unixtime verses unixmillis or strings verses numbers etc that I added the extra log line to see what values are being used in the sum to decide whether to publish or not.
I also wanted to see the "Publishing to ..." lines

Paul-Reed, 2018-07-05:
> Just to confirm, your data is coming into emoncms via http correct?

Nope, all data coming in via RF - emonTX, emonTH.

I've checked the emoncms log, and cannot see any of the entries you've mentioned.

I'll look again tomorrow.... and update

pb66, 2018-07-05:
> Nope, all data coming in via RF - emonTX, emonTH.

That's how the data reaches emonhub (I assume your using emonhub still), emoncms can only get input data via http(s) or mqtt. To word my question another way, am I right to assume you are not using MQTT to input data to emoncms directly using phpmqtt_input.php?

> I've checked the emoncms log, and cannot see any of the entries you've mentioned.

What log level do you have set in settings.php? info is level 1.

> I'll look again tomorrow.... and update

That would be great, cheers Paul.

borpin, 2018-07-06:
@pb66 
> It has just occurred to me this limit will be of no use with mqtt inputs as when the broker cannot deliver it will buffer the data and then a flood of non-timestamped inputs will all end up with a timestamp of now, so any publish to mqtt derived from those inputs will have a "now" timestamp too.

That was what I was trying to explain.

>Luckily in most instances users will not generally have mqtt inputs publishing to mqtt, so the unwritten rule of thumb here will have to be mqtt inputs OR publish to mqtt but not both until the inputs get qos0 and/or timestamps.

Do you mean an input received by MQTT will not then be sent out by MQTT?  I both receive data by MQTT (weather) and publish data originating from an EmonTH *to* MQTT.  The data from my EmonHub is sent to EmonCMS via the HTTP API.

>To my knowledge, all emoncms,log timestamps are always UTC (ie GMT) and although the log timestamps are independent of the timestamps used by the inputs and feeds etc, they too should hopefully all be in UTC too. (TZ times should only really get used when displaying data or daily totals).

EmonCMS stores time as a UNIX integer (seconds since the epoch).  Timezones are only ever seen when you convert that integer to a 'local' time.  So right now it is 1530858505 anywhere in the world.  This translates to `2018-07-06T06:28:25+00:00` or `2018-07-07T06:28:25+01:00` in ISO 8601 format or `Friday, 06-Jul-18 06:28:25 UTC` or `Friday, 06-Jul-18 07:28:25 BST` in RFC 2822 format.

The important thing is what the integer is that any (correctly formatted) string converts back to.

Again, have a look at #821 where I did the work for validating an incoming timestamp for both the API (which despite appearances does not take any notice of a time) and MQTT (even did the help page as well 😄 ).

pb66, 2018-07-06:
> Do you mean an input received by MQTT will not then be sent out by MQTT? I both receive data by MQTT (weather) and publish data originating from an EmonTH to MQTT. The data from my EmonHub is sent to EmonCMS via the HTTP API.

No, I mean that un-timestamped data arriving by mqtt cannot have the publish to mqtt processing throttled using the timestamp that didn't exist and therefore has been issued a timestamp of "now" by emoncms as "now" is always greater than "now" - 120secs. So all the data will be published (as it was) because of the lack of timestamp with the input.

Your MQTT data (weather) I assume is timestamped mqtt, if so it isn't a problem, but even if it wasn't timestamped you are unlikely to republish that data using a publish to mqtt process as other devices can subscribe to the same incoming emon/# topic instead. As for the emonTH, since that data is coming from emonhub via http, it has a timestamp, so that can be published to mqtt and the limiter should work fine, again not really a problem there. 

If for example you were using a pulse counter on your elec meter with an emonPi/emonSD, those pulse counts would arrive at emoncms via mqtt with no timestamp. There would be no point to another device subscribing to that "pre-emoncms" emon/# topic as the pulse count has not been processed yet, so a user would most likely prefer to process that pulse count to a kWh feed and publish that to mqtt. In this instance a backlog of 1000 un-timestamped pulse counts would result in 1000 publish to mqtt's all of the same timestamp (within a sec or 2) which obviously cannot be throttled using time as they will all appear to be less than 120sec old. In this instance the quick fix (to avoid the publish to mqtt overloads) is to switch to using http between emonhub and emoncms so that any one input is not being posted via un-timestamped mqtt and also then being published to mqtt. That's where the "one or other not both" rule applies, but that is only to avoid the "publish to mqtt overloading" position we are already in, this is not a new issue, I am simply saying the proposed "120s limit" fix, will not impact these instances, it does not add any further issue, it simply cannot fix this particular scenario, but using http instead is a quick fix until timestamped mqtt is eventually sorted out.

The point being made about the timezones was that emoncms doesn't use a timezone when writing messages to the emoncms.log file like emonhub does. The emoncms.log "looks" like it's local time because it is formatted as a time and date, but it is not, it's UTC ie unixtime shown as a date/time but with no timezone element. Looking at Paul's Mosquitto logs, they are a unix timestamp not a date/time string, it is easy to expect those to be not local time.

borpin, 2018-07-06:
Hi Paul, I think we are violently agreeing with each other 😄 .

>is to switch to using http between emonhub and emoncms

I'm still at a loss to understand why MQTT was ever adopted for this data transfer.

Paul-Reed, 2018-07-06:
Hi Paul, just re-run the test again, and still the log line did not print.
Log level is already set to info.
I think it may be better to call a halt and wait until you have a test bed, as I'm just starting a new project, and need to work on that.

pb66, 2018-07-10:
No problem @Paul-Reed, I was struggling to find time for this myself. It's unlikely I will set up something to test this on any time soon.

Since @emrysr and @TrystanLea seem to have moved on to something else for now, we can pick this up another time I guess, there's no point in pursuing this further without their input.

Paul-Reed, 2018-07-10:
@emrysr are you still working on this?
It seems sort of half done at the moment as the client now remains connected to the broker (great!), but it's left a situation where if the broker is unavailable for whatever reason, potentially a large number of messages will be cached, and later sent to, and swamping the broker.
This may result in DoS for other MQTT messages and of course those messages will be worthless as they have no timestamp.
It shouldn't affect me personally, as both client & broker are on the same Pi, but could cause problems for others.

TrystanLea, 2018-07-10:
@emrysr and I have been discussing this again today and came up with the idea to use a redis hash rather than queue so that only the last value is published, here's the proposed implementation, testing here it looks like it works well: https://github.com/emoncms/emoncms/pull/914/files

TrystanLea, 2018-07-10:
@emrysr has created an issue for the phpmqtt_input script name here https://github.com/emoncms/emoncms/issues/915

pb66, 2018-07-10:
> so that only the last value is published,

If you read my notes above, this cannot work. "Last value" is only relative to a moving point in time, If I post  1000 frames in 4 batches of 250, that's at least 4 "last values", assuming the frames came from a emonTh (for example) the 4 "last values" would be for approximately 12, 8, 4 and 0 hours ago, but in actual fact even if the 1000 frames were delivered in one hit, the "publish to mqtt" process would be called for each one and therefore each one would be "the last value" until the next one is processed. If the "last one" is relative to the point in time that the mqtt_input script checks it, then again, it could assess what is "the last value" before all the frames are processed.

Plus, if you have (for example) mqtt inputs and http inputs, a mqtt outage could cause a mqtt queue to form and when it reconnects, that backlog gets processed whilst an upto date http input arrives, which also has a publish to mqtt process. Does this other process to mqtt value get discarded with the earlier queued mqtt values if it isn't the last value?

Can we assume from this that you are only interested in publishing QoS 0 "current status" data and no timestamped version will be implemented?

Looking at the code, it looks as if using a hash rather than a queue would have the same effect as simply just popping the last value in the queue (rpop) and resetting it since the hash simply overwrites any previously "publish to mqtt" data that has not yet been published (regardless of whether that data is younger or not).

TrystanLea, 2018-07-10:
> Can we assume from this that you are only interested in publishing QoS 0 "current status" data and no timestamped version will be implemented?

yes that is correct.

> it looks as if using a hash rather than a queue would have the same effect as simply just popping the last value in the queue (rpop) and resetting it since the hash simply overwrites any previously "publish to mqtt" data that has not yet been published

yes that is the desired behaviour, and I think the solution we now have is nice and simple. Im not sure how rpop and reset would work, if its the same then great.

pb66, 2018-07-10:
Going back to @Paul-Reed's example

[image]

How will using a redis hash know to publish the last 3 values not just THE latest value? The point at which the "latest value" is taken and published is controlled by the mqtt_input.php loop, not by the input processing.

The biggest issue with the mqtt input implementation is the lack of provision for timestamps, is it wise to follow that path with the publishing too?

How will a (for example) EV charger know if the current value for excess PV is current or hours old? QoS 0 publishing will mean that when a new device subscribes to the topic published by emoncms, it must wait until there is a value published, and then it will assume that to be "now" even if it is from the middle of a partially processed 2 day backlog.

The hash method effectively blocks all other publish to mqtt processed values whilst any input backlogs are processed.

TrystanLea, 2018-07-10:
I've just re-read the above thread again and realised I missed a few things. I see your point @pb66 about the potentially old entries, even if not a long queue of 1000s of similar updates and your proposed implementation that includes a check for the timestamp being less than 2mins.

I think we could add this with better results to the redis hash approach, assuming:

1) the intended design of the publish_to_mqtt process is for real-time data and qos0. publish_to_mqtt is not designed for use say during upload of historic data via the input bulk api.  

2) therefore passing timestamped data to the subscriber is not a design requirement.

3) the subscriber will only want to know the latest value of say a power feed or kwh feed not all 1000 values queued for the last hour, or even 10 values queued for the last 2 mins.

We could add a last updated timestamp to a redis key that is used alongside the redis hash or perhaps even use expire: https://redis.io/commands/expire

---

Backing up a step, how are people using power_to_mqtt process? I don’t actually use it myself. If its used say for posting data on to another service for data storage, a full queue of timestamped data might make sense, assuming that the service cant subscribe to emonhub directly as its making use of emoncms input processing? Alternatively if its used for real-time event based control, then only the latest value within a defined period is going to be useful and historic queued data might cause the item being controlled to suddenly switch on and off really fast for a period of time...

Paul-Reed, 2018-07-10:
I use it for passing to, and storing data in node-red. I can then integrate the data with other IoT sensors, plus weather data, and access it via a full node-red dashboard menu.
So yes timestamped data would be nice, but not essential for me personally as both Client & Broker are on same Pi.
If you don't implement timestamped data, redis 'expire' sounds like a good solution.

[image]

pb66, 2018-07-10:
> > How will using a redis hash know to publish the last 3 values not just THE latest value?
> 
> Im not sure that I understand this question, what’s the distinction between last and latest?

Sorry, the change in terms wasn't intended, the significant bit was the "THE" as in singular last value. ie

> How will using a redis hash know to publish the last 3 values not just **_THE_** last value?

In @Paul-Reed's example how will the redis hash implementation know to publish all three "publish to mqtt" processes? The input_mqtt is spinning it's own independent loop that isn't synchronized with when the inputs are processed.

Every "publish to mqtt" value **_IS_** the "last value" until the next one, there is no way to know which is actually a real "last value" in the context needed for this to work.

What will trigger the actual publishing of each of the 3 "publish to mqtt" processes in @Paul-Reed's example? Since every "publish to mqtt" value is just over written by the next, it just means a random value will be deemed the "last value" at whatever time the redis hash is read by the mqtt_input loop.

In Paul's example, if lucky it might even seem to work because he doesn't have much in the way of mqtt inputs so the looping may be fast enough to seem correct, but that will fall over when there is a backlog or more mqtt inputs or the broker gets restarted etc etc. 

At least by using a queue at the process end, the MQTT script (and the user?) have the options for publishing timestamped data or to process the queue. By only ever making the single last "publish to mqtt" value available regardless of how many publish to mqtt processes you have is seriously flawed.

Another example, what happens if a single "power" input has 4 publish to mqtt processes, 1st it publishes that power value to `emon/home/power`, then after multiplying that by a unit price it publishes that to `emon/home/costperhour`, then it goes through some power to kwh processes and the results get published to `emon/home/energytoday` and `emon/home/costtoday`. Only the `emon/home/costtoday` will probably ever get published as it's the "last value" to be put to the redis queue/hash. There is nothing to tell the mqtt_input to process and publish the queue/hash inbetween each process being processed by the input module.

> Agreed there is an issue with not including timestamps as you describe but the intention here is to fix the issue that would result from adding a queue without timestamps. E.g 1000s of mqtt publish calls where it should only publish the last value (if that, given that the last value could also be old, but its better than 1000 old values)

That is setting the bar really low. As I previously explained, only a very small user set would continue to see a problem when republishing a non-timestamped value, eg a pulse-count. This "fix" pretty much renders the "publish to mqtt" process useless for everyone. Even if we don't actually provide a timestamped solution right now, it would still not be wise to exclude the possibility in the future for the sake of another quick fix.

I would much prefer to see a small handful of values that span 2mins than see the last value of each batch of processed values, there is little change in 120 secs, but posting a backlog of 1000's of PV values (for example) could result in it appearing like there's 4kW's of excess solar being produced in the middle of the night if the mqtt_input loop deems a value as the "last value" before the processing has actually caught up, I know which compromise I would prefer. Especially as all my good "publish to mqtt" data from other nodes will also get discarded with the bulk of the backlog.

>the intended design of the publish_to_mqtt process is for real-time data and qos0. publish_to_mqtt is not designed for use say during upload of historic data via the input bulk api.

No. The "publish to mqtt" was a quick fix stop gap until a proper mqtt implementation could be considered, what you're now saying is that because that was all that was possible at that time, that is all we could possibly want now we are trying to improve it, that is not the case.

> therefore passing timestamped data to the subscriber is not a design requirement.

For who? That is the same assumption/choice made for the mqtt input, that is the single biggest shortfall of the emoncms mqtt implementation, there is no reason to limit ourselves to ONLY non-timestamped data. Especially after discovering what a headache making that same mistake previously has been.

> the subscriber will only want to know the latest value of say a power feed or kwh feed not all 1000 values queued for the last hour, or even 10 values queued for the last 2 mins.

How could we possibly know that? We would simply be imposing those limitations through bad design choices, the flexibility should be available to explore further uses and applications, not limited by what we currently perceive might be wanted.

> how are people using power_to_mqtt process? 

Or do you mean "how _could_ mqtt output from emoncms be utilized if implemented well"?

> I don’t actually use it myself.

Nor I, because the mqtt in emoncms is really limiting, erratic and unreliable. I would really like to use mqtt in the future and I'm willing to help with the dev work, but if we are going to use the current implementation as a model for future development, I guess that is unlikely to change anytime soon.

TrystanLea, 2018-07-11:
@Paul-Reed if we where passing timestamped data for storage in other services such as node-red would you wish to have history buffered? e.g potentially thousands of updates or would you prefer to only receive the last value for each topic but with a timestamp e.g:

    "publish_to_mqtt":{
        "emon/home/power":{"time":UNIXTIME,"value":100},
        "emon/home/costperhour":{"time":UNIXTIME,"value":200},
        "emon/home/energytoday":{"time":UNIXTIME,"value":300},
        "emon/home/costtoday":{"time":UNIXTIME,"value":400}
    }

Paul-Reed, 2018-07-11:
>  if we where passing timestamped data for storage in other services such as node-red would you wish to have history buffered?

For me, no it's not necessary, the last value would be sufficient, although other users I'm sure will have a contrary view, especially those who are using a remote broker.
Buffering will introduce a number of challenges, for example, what if the broker is unavailable for a lengthy period - how large do you allow the buffer to get? and then what...

Brian @borpin - have you a view on this?

pb66, 2018-07-12:
@TrystanLea - You are right I didn't twig it was a "last value **per topic**" so there should indeed be no problem with multiple Publish to MQTT processes in the same input or recent values from up to date inputs getting lost in the backlog of another input.

I like the "per topic" part but still think the hash is not the best way forward as it excludes any possibility of historic processes being published. A queue (per topic) means the mqtt_input script gets to decide depending on user settings, eg QoS0 just publish the last value, much like you suggest with the hash method whilst QoS2 would publish all values with a timestamp.

I think that my originally proposed idea could be improved by letting the user select a QoS level along side each topic in the publish to mqtt process and if that info was added to the queue too, then the publishing code could handle each topic queue with it's own QoS level rather than a global setting. 

I will look closer at it again, but I'm still convinced the queue is the way forward, the "hash like option" could just get the current queue length and pop values off the queue in a loop until the index matches the queue length at which point it publishes and exits the loop.

This is only an problem because the emoncms mqtt input were designed without timestamps, that is the root of the issue, since changing the input MQTT structure is a big job and we would also need to still provide backward compatability and support the current format, perhaps we should just swap to using a QoS 0 for the current non-timestamped formats. But regardless of that, we really shouldn't let the "known shortfall" of the mqtt input define the way forward.

The "hash method" basically says no user can ever have historical and/or timestamped MQTT originating **from** emoncms because the mqtt **input** has no timestamps, and that applies whether you happen to be using the mqtt input or not. Can we please implement a publishing solution using what we have learnt from the mqtt input and perhaps assume the mqtt input might get fixed/improved/replaced at some time in the future?

borpin, 2018-07-14:
I think the hash is a good solution where you are not trying to accumulate data, you just want to know what the latest 'status' value is. It also works for differing publish rates etc.

There is a use case for a buffered output, but I think that should be timestamped (else makes little sense). I'd suggest the buffer length is configured.

My preference is therefore 2 different ways of creating & loading REDIS depending on whether you simply want a non-buffered 'status' or if you want a buffered output. I suspect 2 different mqtt process options is not too hard to do.

What you do about QOS needs a little more thought as does use of the retain flag and persistent connections.
.
>since changing the input MQTT structure is a big job

I don't agree. I've basically done it on the previously mentioned PR. You just check and see if there is a time specified, if not do as before.

TrystanLea, 2018-07-15:
I've merged the hash method in and kept the queue code in place in phpmqtt_input

https://github.com/emoncms/emoncms/commit/f54dc6166807abe17802d6de93e66dba86a8fe65
https://github.com/emoncms/emoncms/commit/ff9cb6fe50f8cf662e75801b04d242d25c5425a3

and added support to send the timestamp if the queue method is used
https://github.com/emoncms/emoncms/commit/415715b1acf9a951f0bc887f2edcb1fddfe168d2

Question is still open regarding adding a timestamp to the redis hash option

TrystanLea, 2018-07-16:
> I think time for the hash should be configurable and default to 'as is' data format. It could be useful and if configurable, it can be left to the user.

This would need to be a configuration in settings.php I guess $publish_to_mqtt_timestamp = true; ?

> Oh and the base topic for publishing should be different to the subscription base topic (not sure if it is or not).

The user sets the base topic when adding the process.

Interesting idea to use a hierarchical structure!

closed 2018-08-22 by TrystanLea