emoncms/emoncms #909: EmonSD incompatible with Zero or 3B+

borpin, 2018-07-04:
The EmonSD image needs to be updated.  I know there was a recent update to Raspbian so any beta image should probably be moved to the latest release.

At the same time the [Build Guide](https://github.com/emoncms/emoncms/tree/master/docs/RaspberryPi) should probably be checked for validity.

glynhudson, 2018-07-06:
We have a beta image which is working on Stretch (tested on 3B+): https://github.com/openenergymonitor/emonpi/wiki/emonSD-pre-built-SD-card-Download-&-Change-Log#emonsd-13jun18-beta

There are currently some issues with WiFi that need further testing. I'm currently away from the office for another couple of weeks. Getting this up and running will be a priority when I'm back.

borpin, 2018-07-06:
I wonder if the recent update to Raspbian might help as a new base image.

IvanCaregnato, 2018-07-25:
no way to make it work on my old raspberry 2B, so I've used the previous emonSD oct'17 image

TrystanLea, 2019-01-17:
Latest image compatible with 3B+ but perhaps not Pi Zero?

closed 2019-07-11 by TrystanLea