emoncms/emoncms #940: Virtual Feeds to store permanently for apps.

danbates2, 2018-07-30:
Opening discussion here of the possibility to enhance the myelectric app by making virtual feeds available.
Virtual feeds presently are real-time input processors, and thus don't store data. Myelectric realtime power doesn't need data storage, so a virtual feed theoretically isn't a problem for power.. for historical data though..
Perhaps an option could be selected during virtual feeds creation, to store data to be used by, for example, myelectric graphs.
The other approach would be for myelectric to run the 'virtual processors' on historical input processor feed data to arrive at the same information.

chaveiro, 2018-07-31:
For the matter of consuming feeds, Virtual feeds are like any other feed, even with historical data, assuming that the process list of the virtual feed source other 'real' feeds.

danbates2, 2018-07-31:
Understood. I notice the processing time required to look at historical virtual data is quite long.
I'm actual desire more simply a realtime virutal value to be able to be fed to myelectric app.

relft, 2020-06-07:
have you solved? how are you using virtual feed on app, for example mysolar?

danbates2, 2020-06-07:
Not solved.
there was a way to test this. I'll check now for you.

To change the feed type in mysolar for example open
emoncms/Modules/app/apps/OpenEnergyMonitor/mysolarpv.php
see lines 166 onwards.
Change engine id from 5 to 7 for virtual feeds. It wasn't working well for me before.

Engines and their id number are listed in settings.php, or defined for real enum.php line 19 onwards.

relft, 2020-06-08:
Hi, first thanks
I don't really understand if it is supported and working...

danbates2, 2020-06-08:
Unfortunately it is not supported.

relft, 2020-06-08:
And changes thst you supported onprevious messages are working or not?
Thanks

danbates2, 2020-06-08:
Not working. Perhaps in the future.

TrystanLea, 2023-04-06:
Cheers. This has now been fixed, part of the work done here a couple of years ago: https://github.com/emoncms/emoncms/commit/f19d07226b64804f401336de8bf5eb3b27905c5e

closed 2023-04-06 by TrystanLea