emoncms/emoncms #951: Loading apps page.

danbates2, 2018-08-04:
When I first arrive at page 'Apps' from the top navigation bar, it is unclear which page is being loaded, also I can't seem to prioritise which app is loaded first. A simple highlight in the left 'apps navigation bar' would show what app is being loaded and is being worked on/configured.
When pressing 'Apps' from the top navigation bar, possibly also it's a good idea to have no app load, unless a particular app has been 'starred' like in dashboards.

TrystanLea, 2018-08-14:
Thanks Dan, Something like? 
[image]

danbates2, 2018-08-14:
Exactly :)
👍

danbates2, 2018-08-14:
An excellent shade for my screen here.. and just looked at on mobile and it's clear there too.

danbates2, 2018-08-19:
close issue?

closed 2018-08-19 by danbates2