emoncms/emoncms #952: Certbot let's encrypt does not work with .htacess

biva, 2018-08-06:
Hello,
During installation, I tried to activate https with letsencrypt. There was an issue (403 forbidden) during the certbot process(`sudo certbot --authenticator webroot --installer apache`).
I solved it by deactivating the .htaccess (`sudo mv .htaccess .htaccess_bkp`) and re-enabling it after the certbot process was done.
Do you also experiment this?

Config: Debian 9.5, Apache

takkaria, 2018-08-07:
Cerbot generates files in `.well-known` that they need to be able to access remotely.  At the moment this is not allowed.  To fix this you need to add a line like the following to .htaccess:

```
 RewriteCond %{REQUEST_URI} !=/.well-known/.*$
```

After the robots.txt line in the `<IfModule mod_rewrite.c>` block.

biva, 2018-08-07:
Thanks @takkaria : shouldn't it be added in the repository? What about after certbot has done its job, does it create a security problem to keep this new line in .htaccess?

takkaria, 2018-08-07:
Yes it probably should!  I will submit a pull request.

The idea behind Let's Encrypt is that you set up a system to renew the certificate automatically every 12 weeks, so if you don't have that line in .htaccess you won't be able to do that.  There is no security risk in allowing access to `.well-known`, as far as I know.

biva, 2018-08-08:
@takkaria : should I close this issue or keep it open until it's merged in the main project?

TrystanLea, 2018-08-12:
Merged thanks

closed 2018-08-12 by TrystanLea

biva, 2018-10-07:
Hello,
I just tried to renew my certificate with Let's Encrypt: `sudo certbot renew` and the following error appeared:

<code><pre>Domain: emoncms.mysite.com
   Type:   unauthorized
   Detail: Invalid response from
   http://emoncms.mysite.com/.well-known/acme-challenge/jfzeoifjegoirjgzoigjJOIEJFZOIFJEOFIOJEfJEOIfJmfeiJ:
   "<!DOCTYPE HTML PUBLIC \"-//IETF//DTD HTML
   2.0//EN\">\n<html><head>\n<title>403
   Forbidden</title>\n</head><body>\n<h1>Forbidden</h1>\n<p"
</pre></code>

I removed `.htaccess` and it solved the issue. So I think the patch is not enough :(

takkaria, 2018-10-11:
I can help you out here!  In this section of .htaccess:

https://github.com/emoncms/emoncms/blob/master/.htaccess#L176

can you add the line

```
  RewriteCond %{REQUEST_URI} !^/.well-known/.$
```

above the RewriteRule line and tell me if it works then?

takkaria, 2018-10-11:
If you run `sudo certbot renew --dry-run --test-cert` then you can test if your setup works or not even when you don't need to renew.

And yes, I'm aware that line is in a different section too - I think it needs to be in both :)

biva, 2018-10-30:
Hello, I included the line, but I have this error:

 - The following errors were reported by the server:

   Domain: emoncms.mydomain.com
   Type:   connection
   Detail: Fetching
   http://emoncms.mydomain.com/.well-known/acme-challenge/2G_P9i3PXu1LVvIWNrBdYVUfg0t93fLZnyHL3cw2BK8:
   Connection refused

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A record(s) for that domain
   contain(s) the right IP address. Additionally, please check that
   your computer has a publicly routable IP address and that no
   firewalls are preventing the server from communicating with the
   client. If you're using the webroot plugin, you should also verify
   that you are serving files from the webroot path you provided.

Should I restart something before doing the test? I just restarted Apache before renewing the certificates.

takkaria, 2018-11-01:
I have no idea what is happening here I'm afraid!  Could you give more detail about what command(s) you ran to get this error?  I assume you wrote `mydomain.com` instead of your actual domain above?

biva, 2018-11-01:
Sure, you're right for mydomain.com.

I ran `sudo certbot renew --dry-run --test-cert`

takkaria, 2018-11-04:
And can you connect to it yourself?  what does `curl -I emoncms.mydomain.com` show?

biva, 2019-06-01:
@takkaria sorry for the delay of my answer, but I'm back at this issue now. I made a fresh install of the last version of emonCMS (Version: 10.0.2). I tried `sudo certbot renew --dry-run --test-cert --cert-name emoncms.mydomain.com` again and the result is still an issue:

```text
Processing /etc/letsencrypt/renewal/emoncms.mydomain.com.conf
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Cert is due for renewal, auto-renewing...
Plugins selected: Authenticator webroot, Installer apache
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for emoncms.mydomain.com
Waiting for verification...
Cleaning up challenges
Attempting to renew cert (emoncms.mydomain.com) from 
/etc/letsencrypt/renewal/emoncms.mydomain.com.conf 
produced an unexpected error: Failed authorization procedure. emoncms.mydomain.com 
(http-01): urn:ietf:params:acme:error:unauthorized :: The client lacks sufficient authorization :: 
Invalid response from 
https://emoncms.mydomain.com/.well-known/acme-challenge/jPOdiPOIFDPidfIfDfjPoijdfiJfdifjJD8DFDKl9kdlskf5fds [95.54.XX.XX]: 
"<!DOCTYPE HTML PUBLIC \"-//IETF//DTD HTML 2.0//EN\">\n<html>
<head>\n<title>403 Forbidden</title>\n</head>
<body>\n<h1>Forbidden</h1>\n<p". 
Skipping.
```
And still the same error message as in https://github.com/emoncms/emoncms/issues/952#issuecomment-434407986

All other subdomain on my server are working.

`curl -I emoncms.mydomain.com` shows:
```text
HTTP/1.1 301 Moved Permanently
Date: Sat, 01 Jun 2019 00:00:55 GMT
Server: Apache/2.4.25 (Debian)
Location: https://emoncms.mydomain.com/
Content-Type: text/html; charset=iso-8859-1
```

Isn't it a problem of access to the directory were emoncms is installed? For the moment, the owner of this directory `/var/www/html/emoncms/` is me (my username). It's not `www-data`. Could it be the reason?

I can access http://emoncms.mydomain.com from the web and use emoncms without problem (login, ...).

takkaria, 2019-06-01:
I'm looking at my fix now and I'm surprised, the regex is obviously wrong. Try instead:

```
 RewriteCond %{REQUEST_URI} !^/?\.well-known/.*$
```

biva, 2019-06-03:
Thanks @takkaria : I thank you for your quick answer. I modified it, but still the same issue :(

For information, here is my `/etc/apache2/sites-available/emoncms.conf`
```text
<VirtualHost *:80>
    ServerName emoncms.mydomain.com
    DocumentRoot /var/www/html/emoncms

<Directory /var/www/html/emoncms>
    Options +FollowSymlinks
    AllowOverride All
    DirectoryIndex index.php
    Order allow,deny
    allow from all
</Directory>

RewriteEngine on
RewriteCond %{SERVER_NAME} =emoncms.mydomain.com
RewriteRule ^ https://%{SERVER_NAME}%{REQUEST_URI} [END,QSA,R=permanent]
</VirtualHost>
```

And my `/etc/apache2/sites-available/emoncms-le-ssl.conf`:
```text
<IfModule mod_ssl.c>
<VirtualHost *:443>
    ServerName emoncms.mydomain.com
    DocumentRoot /var/www/html/emoncms

<Directory /var/www/html/emoncms>
    Options +FollowSymlinks
    AllowOverride All
    DirectoryIndex index.php
    Order allow,deny
    allow from all
</Directory>

<Directory /var/www/html/emoncms/.well-known/acme-challenge>
     Order Deny,Allow
     Allow from All
</Directory>

SSLCertificateFile /etc/letsencrypt/live/emoncms.mydomain.com/fullchain.pem
SSLCertificateKeyFile /etc/letsencrypt/live/emoncms.mydomain.com/privkey.pem
Include /etc/letsencrypt/options-ssl-apache.conf
</VirtualHost>

</IfModule>
```

takkaria, 2019-06-03:
If you put a file in /var/www/html/emoncms/.well-known/acme-challenge called 'test', what happens if you try to fetch it in a web browser?

biva, 2019-06-04:
Same: Forbidden :(

```text
Forbidden

You don't have permission to access /.well-known/acme-challenge/test on this server.
Apache/2.4.25 (Debian) Server at emoncms.mydomain.com Port 443
```

biva, 2019-06-04:
For information, if I remove `.htaccess` then I can access `/.well-known/acme-challenge/test` and `sudo certbot renew --dry-run --test-cert --cert-name emoncms.mydomain.com` works as well.

So the problem really comes from the `.htaccess` I guess :(

takkaria, 2019-06-04:
There's a section lower down in that file:

```
# Block access to hidden files and directories.
# This includes directories used by version control systems such as Git and SVN.

<IfModule mod_rewrite.c>
    RewriteCond %{SCRIPT_FILENAME} -d [OR]
    RewriteCond %{SCRIPT_FILENAME} -f
    RewriteRule "(^|/)\." - [F]
</IfModule>
```

If you delete it what happens?

biva, 2019-06-04:
It works :)

What's the impact of removing these lines?

takkaria, 2019-06-05:
It's unclear to me that they have any impact at all because of sections
above. I'll submit a pull request for testing though.

biva, 2019-06-05:
OK, and should I leave `RewriteCond %{REQUEST_URI} !^/?\.well-known/.*$`?

biva, 2021-04-26:
@takkaria I'm working on this issue again, do you know more since our last communication?

In https://github.com/emoncms/emoncms/blob/master/.htaccess, I still see the block that you advise to delete in https://github.com/emoncms/emoncms/issues/952#issuecomment-498568594

biva, 2021-04-27:
New issue created to follow up: https://github.com/emoncms/emoncms/issues/1684