emoncms/emoncms #978: Server info incorrectly reported when using the copy to clipboard button

pb66, 2018-08-18:
See https://community.openenergymonitor.org/t/rpi-3b-getting-hot-redis-high-cpu-usage/8263?u=pb66

Whilst the admin page says "0 feed points pending write" in the Emoncms/buffer section

[image]

the information copied by the "copy to clipboard" button reports "loading"

[image]

I suspect one or the other is telling porkies

TrystanLea, 2019-01-17:
Note: Still present in latest master 17 jan 19

TrystanLea, 2019-07-11:
This was fixed in a more recent version I believe

closed 2019-07-11 by TrystanLea