emoncms/emoncms #990: Save user preference for expanded or contracted input/feed lists

TrystanLea, 2018-08-22:
There is a differing preference for expanded/contracted input/feeds lists, I wonder if we could save this preference in the new user preferences feature in the user module?

emrysr, 2018-09-18:
works well with issue #1021. as you swipe from one to the other the state of the accordion/dropdowns are remembered across page loads

pb66, 2018-10-02:
@TrystanLea and @emrysr 

Will this persist across logins or devices?

The preference to have all collapsed on first view (or not) is very different to remembering the current choice whilst swiping pages.

I think there still needs to be a default user setting, I would prefer that new pages always opened all collapsed, even if I did leave a panel or 2 open when I last visited.