emoncms/emoncms #991: Review publish_to_mqtt format, option to use timestamps

TrystanLea, 2018-08-22:
See end of discussion on https://github.com/emoncms/emoncms/issues/903

Should I add a timestamp even in the redis hash option? e.g:

    "publish_to_mqtt":{
        "emon/home/power":{"time":UNIXTIME,"value":100},
        "emon/home/costperhour":{"time":UNIXTIME,"value":200},
        "emon/home/energytoday":{"time":UNIXTIME,"value":300},
        "emon/home/costtoday":{"time":UNIXTIME,"value":400}
    }

The danger is that it will likely break things for existing users! Alternatively it could be a settings.php setting..

@borpin replied

> I think time for the hash should be configurable and default to 'as is' data format. It could be useful and if configurable, it can be left to the user.
>
> Oh and the base topic for publishing should be different to the subscription base topic (not sure if it is or not).
>
> I floated on the forum that we should change how the settings are loaded to a hierarchical structure. I have now raised issue #923. If this number of changes are being made to the settings file it might be an opportune moment

This would need to be a configuration in settings.php I guess $publish_to_mqtt_timestamp = true; ?

Need to review hierarchical structure proposal.

borpin, 2018-08-22:
I think it needs to be an option as it will break too many things - the current format has been in place for quite a while.

TrystanLea, 2018-08-29:
settings.php setting name suggestion @emrysr:

    // include timestamp in messages from publish_to_mqtt process:
    // example: "emon/home/power":{"time":UNIXTIME,"value":100}
    $publish_to_mqtt_timestamp = true;

karlp, 2018-10-28:
it's json, so adding a key is transparent? what's the backward compatibility concern?

borpin, 2018-10-29:
> it's json, so adding a key is transparent? what's the backward compatibility concern?

Currently it is a single value.  Any processing will expect a single value, sending it as a JSON will likely break the users implementation.

TrystanLea, 2025-07-16:
Closing as low interest? Feel free to open in future.

borpin, 2025-07-16:
It is part of the JSON spec, just not for individual feeds. @TrystanLea

TrystanLea, 2025-07-16:
Thanks @borpin ah are you thinking inputs created using an external time parameter should then have that time passed through to publish via mqtt?

borpin, 2025-07-17:
> Thanks [@borpin](https://github.com/borpin) ah are you thinking inputs created using an external time parameter should then have that time passed through to publish via mqtt?

Oh, sorry, this is publish *from* Input Processing - ignore 😄