emoncms/emoncms #1000: Beta features - feed unit displaying 'undefined'

Paul-Reed, 2018-08-23:
I’m assuming that the ‘feed unit’ is not mandatory, and is at the discretion of the user if they wish to assign a ‘unit’ to their feed or not.
That being the case, could you consider removing the ‘undefined’ text please.

[image]

So that users just see;
23.2°C
33.7
2.9

TrystanLea, 2018-08-23:
Thanks @Paul-Reed That's strange I thought this was only an issue affecting emoncms.org, I've copied the fix across, now in master: https://github.com/emoncms/emoncms/commit/a1a427971eb737cb0a02155495f4daf5fa1f861e

Paul-Reed, 2018-08-23:
Yes, that's now sorted.

Thanks

closed 2018-08-23 by Paul-Reed