emoncms/emoncms #997: Trying to read value from feed to use in textbox

CambridgeEP, 2018-08-23:
Hi,
This may be a stupid question, but I am trying to read a value from my feeds to insert into a textbox on my dashboard. 

I have tried:
<iframe src=https://emoncms.org/feed/value.json?id=370069 </iframe>
But with this I can only get the value in "" and cannot edit the text format.

Also looking at reading out of the JSON but have no idea how... This is my attempt:
<a href="<?php echo $path; ?> https://emoncms.org/feed/value.json?id=370069"><?php echo $path; ?> https://emoncms.org/feed/value.json?id=370069</a>

Help appreciated!

Paul-Reed, 2018-08-23:
We prefer general questions such as this to be asked in the forum, instead of raising a git issue.
https://community.openenergymonitor.org
Also, this is the emoncms core repo and not the dashboard repo.

Could you please post in the forum, and I'm sure one of the community experts will be able to assist.

closed 2018-08-23 by Paul-Reed

CambridgeEP, 2018-08-23:
Sorry, new to this, my bad

Paul-Reed, 2018-08-23:
No worries!