emoncms/emoncms #1001: Common engineering units list/lib

pb66, 2018-08-24:
Recently @emrysr  has added a dropdown list of units for use in the feed tables.

It would be nice to see a universal central engineering unit list that could be accessed from various places, eg the graphs module or apps module etc, but more importantly, in the various dashboard widgets eg

[image]

It would be nice to be able to pick "°C" from a dropdown rather than using ALT-0176 and SHIFT-C.

It would allow a single place to maintain the usual units, offer uniformity in how the units are represented and possibly even eventually open up an opportunity for users to "drop-in" additional custom units (to be picked up and added to the stock unit list) via settings.php or maybe their own file, in the theme perhaps.

TrystanLea, 2019-01-17:
Is this issue ok to close? @emrysr

emrysr, 2019-01-17:
the `/emoncms/Lib/units.php` already returns a JSON object with common units.

however, nothing in the dashboard repo makes use of it.

I think that this issue can be closed and new issues can be created in other Modules that make use of these common units