emoncms/emoncms #1004: user defined process quick links

pb66, 2018-08-25:
Extending #926 (already a great improvement, especially if users use the log to feed, power to kWh and +input processes a lot)

[image]

What might be nice is to add a "quick link" checkbox on the far right of any selected process so that users can select and deselect their own favorite quick links

[image]

so the quick link bar can auto populate depending on the processes selected by the user as the most used, they only need to manually select the process once and then tick the box for it to become a quick link.

TrystanLea, 2019-01-17:
Thanks @pb66 this will likely add a lot of additional complexity, the preference will need to be stored stored and retrieved. I would prefer to keep this behaviour simplified for now, but can leave this one up for the future.

pb66, 2019-01-19:
> but can leave this one up for the future.

That's fine, I had anticipated developing some additional process modules so the dropdown might get a bit busy, but although. I do have additional process modules they are not ready for general release yet.

TrystanLea, 2019-07-11:
I don’t think we are going to get a chance to work on this due to limited time resources. Happy to receive a pull request if anyone else wants to try implementing it?

closed 2019-11-25 by TrystanLea