emoncms/emoncms #1006: An anomoly with editrealtimedata timestamps

pb66, 2018-08-28:
See https://community.openenergymonitor.org/t/unearthed-a-quirk-with-editrealtimedata-timestamps/8378?u=pb66

The editrealtimedata functionality has in the past been a bit erratic. Today I noticed a 3 second offset in the datapoint timestamps reported by the editrealtimedata viz. I now suspect the reason it "doesn't work" half the time for me is because I find or search out the erroneous datapoints in graph and therefore use timestamps the editrealtimedata vis cannot find.

On some occasions it has worked fine when manually selecting the spike, but when entering the "correct" details it will not stick, I know think that's because the data point doesn't exist, only editing the value without paying any attention to the timestamp would actually work.

pb66, 2018-08-28:
Some further testing proved the offset to be consistent across any given window of data, But the offset varies depending on the level of zoom or the size of data etc. The same data point's timestamp will vary as you zoom in and out.

pb66, 2018-08-31:
New info, negative timestamp errors/offsets do not allow editing of the selected datapoint. Only positive error will permit successful edits of the target datapoint.

See #1011 and https://community.openenergymonitor.org/t/unearthed-a-quirk-with-editrealtimedata-timestamps/8378/4?u=pb66 for more info.

TrystanLea, 2019-01-17:
There has been recent work on this by @alexandrecuer 

See:

https://github.com/emoncms/emoncms/pull/1147
https://github.com/emoncms/emoncms/pull/1148

can you confirm if the issue is still relevant?

alexandrecuer, 2019-01-17:
Single data modification was not working for phptimeseries but now it should work...the job is done via redis buffer so you have to wait for the feedwriter service to operate....

For phpfina with tight intervals, it is quite complex to modify single values....the process is also using the redis buffer....so you have to wait a little....maybe there is a problem similar to phptimeseries with the limitinterval parameter when you get the datas ...but phpfina get_data() is different than the phptimseries get_data()...i will check at spare time.....

The modification of a datarange (scaling or nan replacement or replacement by absolute values) is now working....it is done directly on the dat file in a single write, without using the redis buffer....from my point of view, it is very enough for work on phpfina feeds

Alex

TrystanLea, 2020-01-20:
This is now fixed see: 
https://github.com/emoncms/emoncms/issues/1477

and

https://github.com/emoncms/emoncms/commit/67419fccbc35eaab401f706da62cb35a656acd8e

closed 2020-01-20 by TrystanLea