emoncms/emoncms #1012: Wrong branch displayed in admin view

Paul-Reed, 2018-09-04:
This was introduced in git pull https://github.com/emoncms/emoncms/pull/994 by @borpin 

The git branch displayed in the admin panel does not reflect the actual branch being used, and could cause confusion....

[image]

borpin, 2018-09-04:
Did you chnage branch?  Is it a cache problem?

The command used is 

```
"git -C " . substr($_SERVER['SCRIPT_FILENAME'], 0,strrpos($_SERVER['SCRIPT_FILENAME'], '/')) . " branch"
```

Is it possible the location variable used is incorrect?

Paul-Reed, 2018-09-04:
I've been on Master branch for at least a month, but have switched branches numerous times in the past.

It would be great if this could be fixed Brian, or if not, revoked, because it's currently providing misleading information, and could influence the support given to users through the forum.

Paul

borpin, 2018-09-04:
Hi @Paul-Reed, could you open the admin window in a private window to confirm if it is a cache problem or not please?

Paul-Reed, 2018-09-04:
It's not a cache problem, I've tried a incognito chrome browser and it shows the same.

Paul-Reed, 2018-09-04:
How do you want that line to look like? just take out the `@exec`?

borpin, 2018-09-04:
@Paul-Reed see my edit above - I know why, just not how to fix it.

borpin, 2018-09-04:
I think perhaps

```git branch --contains HEAD``` might be better than ```git rev-parse --abbrev-ref HEAD```.

I can see the logic of the first command and have no idea what the second is doing 😄.  Sorry need to go and will not get a chance to do a PR until next week.

Paul-Reed, 2018-09-04:
That now retrieves the master branch (which is correct).

You can always sync the remote branches yourself for testing your changes, I seem to recall it's `git fetch`.

Paul

borpin, 2018-09-13:
I didn't do the first PR correctly and only submitted it to my fork. Hopefully this is now a PR for emoncms/master.

glynhudson, 2018-09-20:
Great, nice work #1017 has fixed this issue 👍

closed 2018-09-20 by glynhudson