emoncms/emoncms #1014: Compatibility of emoncms and swiftmailer

pb66, 2018-09-05:
It seems swiftmailer has moved on to v6 which requires PHP v7+ (See [documentation](https://swiftmailer.symfony.com/docs/introduction.html#installation)), that in itself is not necessarily a problem as most platforms have moved onto PHP v7+ now. However, it seems v6 no longer uses `newInstance()` (see the [changelog](https://github.com/swiftmailer/swiftmailer/blob/master/CHANGES#L52)) which emoncms does (see [Email.php](https://github.com/emoncms/emoncms/blob/master/Lib/email.php#L22)).

The [PEAR channel](http://pear.swiftmailer.org/) is still available but no longer supported, so if users install using the emoncms install guide, they will still get v5.1.0 and that seems to work as expected with emoncms, even on PHP7.

The latest versions of swiftmailer are available via composer, we have previously discussed moving to using composer rather than PEAR/PECL but since the redis and mosquitto extensions are not serviced on composer it seemed daft making the switch to catch one lesser used module and lose easy access to 2 more essential extensions.

Perhaps we should put a note in the installer guides that emoncms is only compatible with swiftmailer version 5 not 6 as a temporary measure?

Perhaps if we want or need to make the jump to v6 we should look at revising both the code and the install method, (which will possibly mean using both composer and pear.pecl to get all the modules).

Otherwise if sticking with the v5 for the time being, since it isn't going to get any updates, we could include the swiftmailer v5 lib in the emoncms repo to avoid confusion and ensure the right version is available. If users install swiftmailer v6 elsewhere for other applications, emoncms would then still use the included v5 without any issues.

see https://community.openenergymonitor.org/t/input-process-send-email-is-not-working/8454?u=pb66 (and #1013)

cagabi, 2018-09-05:
> Otherwise if sticking with the v5 for the time being, since it isn't going to get any updates, we could include the swiftmailer v5 lib in the emoncms repo to avoid confusion and ensure the right version is available.
I know it makes the repo bigger in size but I think this is the best option

TrystanLea, 2018-09-20:
My inclination is towards not including, but how much bigger are we talking about? Sticking to v5 for the time being seems sensible, I still need to look into composer.

TrystanLea, 2019-07-11:
Now that we are moving to php7 I think we can close this issue. I haven’t seen it arise as a major issue either for those running older versions of php, we just need to keep in mind that v5 should be used if this is the case.

closed 2019-07-11 by TrystanLea

TrystanLea, 2020-03-17:
Swiftmailer v5.1.0 is no longer available via pecl. Taking a look at what is required for v6 support it looks like there are a number of new dependencies via composer. Updating the code by itself to remote newInstance did not work.

It is however possible to use v5.4.8 via git installation, I have added a note to the effect to the start of the email.php: https://github.com/emoncms/emoncms/commit/8dbeaabdda8f5fb5eb53ff804cb0e39be281ed44

    cd /opt/emoncms/modules
    git clone https://github.com/swiftmailer/swiftmailer.git
    cd swiftmailer
    git checkout v5.4.8

This is an interim fix for now.