emoncms/emoncms #1016: MySQL database setup for user registration.

danbates2, 2018-09-12:
I recently went through an install and the problem still exists.
https://github.com/emoncms/emoncms/issues/791

I also had to do the following:
ALTER TABLE users CHANGE preferences preferences TEXT CHARACTER SET utf8 COLLATE utf8_general_ci NULL DEFAULT NULL;

Where is the database setup script?

danbates2, 2018-09-12:
I believe if we drop in :

`ALTER TABLE users CHANGE tags tags TEXT CHARACTER SET utf8 COLLATE utf8_general_ci NULL DEFAULT NULL;`

`ALTER TABLE users CHANGE preferences preferences TEXT CHARACTER SET utf8 COLLATE utf8_general_ci NULL DEFAULT NULL;`

here:
https://github.com/emoncms/emoncms/blob/master/Lib/dbschemasetup.php

this would be a fix?

takkaria, 2018-09-13:
If you just remove 'null'=>true from user_schema.php it should work.  I did submit a pull request that rewrote the database schema code in a more robust way in #930 but no-one's accepted it yet.

danbates2, 2018-09-14:
Thanks. I've just tried that on a fresh install and it hasn't worked.
```
    'tags' => array('type' => 'text', 'default'=>NULL, 'Null'=>true),
    'startingpage' => array('type'=>'varchar(64)', 'default'=>'feed/list'),
    'email_verified' => array('type' => 'int(11)', 'default'=>0),
    'verification_key' => array('type' => 'varchar(64)', 'default'=>''),
    'preferences' => array('type' => 'varchar(255)', 'default'=>NULL, 'Null'=>true)
);
```
For a long shot I tried removing **, 'default'=>NULL,** also.

danbates2, 2018-09-14:
I've also tried your changes, including both your dbschemasetup.php and user_schema.php, fresh install, new database. I get this error.
[image]

danbates2, 2018-09-14:
That worked Perfectly.
I didn't realise originally that I had to delete the original database. I'm new to this.

danbates2, 2018-09-14:
@glynhudson @TrystanLea 
Hi Both, a very simple fix to the problem is to be found in takkaria's comment above.

danbates2, 2018-09-17:
@glynhudson 
Thought to tag you in Glyn, Trys is still away on hols :) Check out this simple fix in the pull request.

closed 2018-09-20 by danbates2