emoncms/emoncms #1020: Readme?

glynhudson, 2018-09-17:
Please add a readme for the Schedule module with install instructions and a brief overview of functions?

TrystanLea, 2018-09-20:
Is this the inbuilt schedule module from @chaveiro rather than demandshaper for OpenEVSE?

glynhudson, 2018-09-20:
Yes, the built-in Schedule module. I was struggling to find any
documentation for how to use it.

pb66, 2018-09-20:
I agree, it would be nice to be able to offer a link to some documentation in this forum thread https://community.openenergymonitor.org/t/how-to-use-schedule-how-it-works/8570?u=pb66, I don't know enough about it to offer any assistance.

Perhaps @chaveiro can offer some help, even if it's just some points to get it started and it can be added too over time?