emoncms/emoncms #1034: Input page : sort order from curl POST

danbates2, 2018-09-22:
I noticed that if I do a curl POST request the sort order of the resultant node's keys are reversed.

[image]
[image]

Would be nice if they were in the order the data string came, for example this would be best:

curl -d "node=testnode&data={respect:123,for:456,women:789}&apikey=" "http://emoncms.localhost/input/post"

Input page view:
|||testnode|||
respect          123
for                  456
women           789

TrystanLea, 2018-09-22:
Thanks Dan, I think the discussion and developments on indexedinputs may solve this one https://community.openenergymonitor.org/t/development-indexed-inputs/8354/35

danbates2, 2018-09-23:
I think it will :)

closed 2018-09-23 by danbates2