emoncms/emoncms #1036: Feed and input page font size and weight

pb66, 2018-09-22:
Each row on the feeds page is quite deep, could we increase the font size or weight for the feed name and value/unit text to improve readability without causing an increase in row size?

Likewise on the input page, the input names and values would benifit from a heavier or larger font too.

See https://community.openenergymonitor.org/t/thoughts-on-the-new-input-and-feed-page-layouts/8583?u=pb66

TrystanLea, 2018-09-22:
Personally I didn’t really notice this as an issue, I think @emrysr experimented with something similar to what you propose but I preferred the way it is currently shown with standard font, it looked cleaner, less clunky in my mind.. This is likely a personal preference thing.. we can certainly try a number of options again

pb66, 2018-09-22:
> preferred the way it is currently shown with standard font, it looked cleaner, less clunky in my mind.. This is likely a personal preference thing

It's not really a "personal preference", I struggle to read it, at best I'm squinting, it's not easy to read or comfortable  to do so and I'm sure there are possibly users with even poorer eyesight than mine out there.

The row size has increased but not the font size so there is room for bigger text without any increase in  size overall. 

If you are considering mobile use too, big and bold is going to be almost essential.

emrysr, 2018-09-23:
The problem with increasing the font size is that it effects the available space horizontally.

A larger font size would reduce the number of columns shown on a mobile device, as the feed title column width increases and takes up the available space.
[screenshot of iPhone5 320x568px with font-size set to 18px] ** not taken from the actual device
[image]

I can increase the font for just larger devices but the "cut off" point would be different for all users based on the feed name.

closed 2019-01-17 by TrystanLea