emoncms/emoncms #1037: Where have the virtual feed edit processlist buttons gone?

pb66, 2018-09-22:
The new feeds page appears to be missing the buttons to open the processlist editor.

(See https://community.openenergymonitor.org/t/thoughts-on-the-new-input-and-feed-page-layouts/8583?u=pb66)

TrystanLea, 2018-09-22:
The button should appear in the top bar if you select the feed, do you see it?
[image]

emrysr, 2018-09-24:
@pb66 - well spotted thanks.
pr #1044

-  Only one feed processlist can be edited. If any other feed type is selected then the button disappears (regardless of the order you select them)  
-  Stops selected feeds from being hidden