emoncms/emoncms #1038: Feeds page buttons at both top and bottom of page

pb66, 2018-09-22:
Should all the butttons be grouped together rather than split between the start and the end of the page?

Style wise the "create new virtual feed" button could be reduced to a "+" button with a more informative hover-over like the new "select all" etc. Likewise the "refresh feed size" could be reduced to refresh symbol. 

(See https://community.openenergymonitor.org/t/thoughts-on-the-new-input-and-feed-page-layouts/8583?u=pb66)

TrystanLea, 2018-09-22:
In a way I think it would be good to keep the virtual feed button separate as its distinctly different functionality to standard feeds.. but can be persuaded

pb66, 2018-09-22:
> In a way I think it would be good to keep the virtual feed button separate as its distinctly different functionality to standard feeds.

Fair comment, but you already have a "edit virtual feed processlist" button at the top and a refresh (all) feed sizes (virtual feeds do not have a size) at the bottom, so there is currently no clear distinction by location.

emrysr, 2018-09-23:
More testing of different devices is required to see what is the smallest device width we support. 
Here's a screenshot of the iphone5 screen size:
[image]

The row of buttons wrap to 2 line and overlap the list. 

Not sure if pr #1040 should be pulled if you support small devices?

pb66, 2018-09-23:
You could reduce by one button by (for example) To be honest on a mobile device I would rather see a burger bun dropdown appear on the top left before the "Feeds" title and have full button descriptions rather than a row of icon only buttons, that would reduce the limit on the functionality available via buttons dur to size restriction, offer users a better idea of what they are pressing as there is no mouse over on the mobile devices and it would give us an extra line or 2 for feed data..

emrysr, 2018-09-24:
thanks @pb66 - I've added to pr #1040 with the above suggestions put in place

closed 2019-01-17 by TrystanLea