emoncms/emoncms #1055: Mobile view doesn't scroll up or down

Paul-Reed, 2018-09-30:
Using current emoncms & device module 'master' git versions, and viewing on Android mobile device.
Both the inputs & feeds pages will not scroll up or down, so you cannot see all of the pages.

pb66, 2018-09-30:
I can confirm this as I noticed this too, i just haven't had time to debug or report as yet.

I also noticed there are no values on either page, unless they are perhaps on the far right and the horizontal scroll is also not working (but I assumed that with the "swipe action" to switch pages the horizontal scroll may not be possible).

If you click a feed and get the data viewer (graph) page the horizontal scroll does work in places, but it is very hit and miss. The vertical scroll is a bit better but not good.

If you navigate to the admin page, you can just about get both the vertical and horizontal scrolls to work if you find the magic zones. for me the very left hand edge of the screen will work the vertical scroll most of the time.

Paul-Reed, 2018-10-19:
There's still an issue with this @emrysr 
I've been working from my Android phone all day today (sigh!), and wanted to check some details from the 'Administration' page, and found that I could only see the left hand side of the page, and couldn't scroll horizontally. Vertical scrolling was OK.
Since the screenshot, I've updated to v9.9.1-beta, and still the same.
[image]

emrysr, 2018-10-19:
@Paul-Reed - thanks for finding this.. I've put a pull in a pull request to fix it.

PR #1057 (2018-10-02): locked the swipe navigation to horizontal only.
this will allow mobile users to still scroll / drag the page vertically

fix #1055
PR #1073 (2018-10-20): removed the dependance on hammer js on views that need it
fix horizontal scrolling issue on admin page
fixes #1055
closed 2018-10-20 by glynhudson