emoncms/emoncms #1060: Consider emoncms using user account TZ rather than browser TZ when viewing data and dashboards etc

pb66, 2018-10-02:
See https://community.openenergymonitor.org/t/timezone-in-emoncms/7569?u=pb66

In short (for example) when we look at a solar plot for any given day it should start to increase AM and tail off PM as a rule. it should not look like the sun rose at 6pm and set again at 6am and peaked at midnight just because it is viewed from Australia. In most cases the data is related to the user account TZ and not the TZ that the data is being viewed from at any given time.

If users in 5 different TZ's view the same data that relates to a single location at the same time, it should appear the same to all users, not be TZ adjusted for each user.

TrystanLea, 2018-10-26:
Its a good idea, I think this would be worth implementing

emrysr, 2019-01-28:
I've implemented a timezone dropdown in the graph module
[image]

https://github.com/emoncms/graph/pull/51

pb66, 2019-07-29:
This issue should not be closed as the feature implemented in the graph module only affects the graph module. We need a system wide fix so that ALL data is presented in the user account TZ.

In addition to this it's worth noting that the use of browser time rather than server time also has it's issues, see https://community.openenergymonitor.org/t/emonpi-gone-mental/11206?u=pb66.

The bottom line is it really is irrelevant where in the world the data is being viewed or what time the client device thinks it might be. The sun still comes up at the same time for the house being monitored, the heating still comes on at the same time and when we look at the feed or inputs pages if the last 10s update was less than 10s ago it should display a number less than 10, even if the client device is an hour slow, the last update was still under 10s ago, not an hour and 10s ago.

camelcamro, 2020-04-17:
still waiting ? - still no solution ?
i don'T undersstand what is here so complicated that api use the time from installed/configured where the server is running. that is nothing special, isn't it ?

borpin, 2020-04-17:
There has been some TZ discussion here https://github.com/emoncms/emoncms/pull/1545

Personally, the issue is actually that I think you want to view the TZ relating to the Feed TZ; what the user or browser TZ is is irrelevant. Take Solar generation data for instance, when looking at it, you want to see the data on power generated related to the local time of that data irrespective of your or the browser TZ. There could be 2 users on a single instance, in different TZs, but they should be looking at the same graph related to the Feed Data.

camelcamro, 2020-04-17:
yes, should be as you wrote .... but that is not the case ... maybe need more patient :( and waiting another 2 years to get a solution ;-(
thx

chaveiro, 2020-04-17:
If you change your computer / phone timezone to the timezone you what would that work? 
For example when abroad my phone switches itself to that territory's timezone so i get local time on the graphs instead of original feeds time at my home. And that is the rational here, you get browser timezone that is equivalent to the device timezone, where the viewer is located.

borpin, 2020-04-18:
> And that is the rational here, you get browser timezone that is equivalent to the device timezone, where the viewer is located.

But that is not logical. If you are looking at the amount of solar generation, recorded by a Feed in the UK, when you are in Aus, it will show max generation at (around) Midnight which is clearly illogical.  You would need to do a manual conversion to see when that really occurred.

Have both means of viewing if you must, but logically there **should** be a way of aligning the feed data with the local time it was actually collected at. Currently that is *assumed* to be the users TZ (and I hate assumptions...).

TrystanLea, 2020-04-19:
It's worth noting that the solution to this whether user based or feed based is not a simple fix as it requires changes to every single visualisation, graph, app on client side as well as changes on the server side to how timezones are handled.

For an example of one place in many that uses browser based timezones to handle preparation of data to be graphed we have calls to JS date.getFullYear etc in the bargraph visualisation here: https://github.com/emoncms/emoncms/blob/master/Modules/vis/visualisations/bargraph.php#L274 

Looks like a lot of changes and testing across the board, in addition to bringing in the timezone settings via additional API calls to each feed in a graph..  Not saying we shouldn’t do it, just that its not a quick fix unfortunately. Just so that we are aware of the scope of work involved.

TrystanLea, 2020-04-19:
I also want to check that we are all on the same page as to what does and what does not work?

- The current implementation works for users who have all their data/devices/browser in a single timezone and access that data within that timezone. E.g Im in Europe/London and my devices are all in my house on the same timezone, my browser is on the same timezone. (This is the design basis of the timezone implementation in emoncms).

- It does not work so well when you travel to different timezones. E.g If I go to France and I try and look at the MyElectric app or MyHeatpump app it will not show the current day, as there is a mismatch between the account timezone and the browser timezone. The rest of the graph works ok. If I quickly switch my browser timezone to the data timezone the current day would reappear (@chaveiro's suggestion).

- It does not work so well if I have say multiple solar sites under a single emoncms account on different timezones and I want to see each graph in the timezone of the recorded data (@borpin's example)

Are there any other timezone related issues that I have not listed?

borpin, 2020-04-19:
> It's worth noting that the solution to this whether user based or feed based is not a simple fix as it requires changes to every single visualisation, graph, app on the client side as well as changes on the server side to how timezones are handled.

I never imagined it would be simple....

> * The current implementation works for users who have all their data/devices/browser in a single timezone and access that data within that timezone.

Agreed

> * It does not work so well when you travel to different timezones.

Agreed

> * It does not work so well if I have say multiple solar sites under a single emoncms account on different timezones

Agreed.

I would add that if you have different users in different timezones on one account it does not work.

Whilst it is understandable that case one was used in the initial design, it is flawed in that it is the local time of the Feed source, that is critical. It will still be stored in the same way (assume a time string is local unless there is a TZ and convert on that basis). If the data is collected at 01:00+10:00 it should always be viewed against that same local datum unless the user asks for the time to be converted.

chaveiro, 2020-04-19:
Let me try to explain more on all the variables:
1 - There is a device timezone (currently is user TZ) that serves to calculate daily accumulation eg Kwh/day feed used by feed processors. This is done server side.

2 - There is viewer timezone (currently its browser TZ). This is done in JavaScript side.

The first variable (feed TZ) cant change after creation or it will corrupt daily accumulations after change  due to midnight shifting.

The second variable (JavaScript side) is used for visualization only and can be what ever you want. Its device TZ here. 

The usecase to have different TZ on the same graphs and dashboars is on the visualization layer (JavaScript) but will not affect daily calculations as that depends on feed TZ.

So if you are on the next day already on the
visualization layer and feed TZ is still on the day before. Your visualization layer has no way to show actual day DATA as the feed date is just one variable per day and it did't switch to the next day yet for the feed timezone.

Feed TZ is where the device is installed.  
The usecase to have daily sums for a timezone different to the device placement is strange to me.

borpin, 2020-04-19:
> Let me try to explain more on all the variables:

You are missing a variable

> There is a device timezone

Yes but this is actually the server (emonCMS) timezone, not the **device** timezone. The **device** is the sensor generating the data at source i.e. the temperature sensor, the CT, the flow meter, which is then added to the Feed.  This is the variable you are missing.

One user might have multiple **devices** (sensors), but not all those devices are necessarily in the same TZ - that is assumption that has been made. Nor are they necessarily in the same TZ as the server (emonCMS for instance).

How that data is then visualised and anchored to the local time it was collected can be adjusted but fundamentally, if you want to know the temperature at midday, where the temperature sensor is in the world, you need to store that temperature sensors TZ with the Feed.

You should not assume the sensors are all in the same TZ as each other, as the user, or the server (although that is exactly what is currently done).

TrystanLea, 2020-04-19:
'Server timezone' has the potential to confuse things I think. At the moment there are two timezones used in emoncms

- **User timezone**, associated with each users account of which there can be many on any one server
- **Browser timezone** (applied on the client side to visualisations) involved in features such as showing the last day of consumption etc..

borpin, 2020-04-19:
> * **User timezone**, associated with each users account of which there can be many on any one server

And which every Feed defaults to regardless of where the data is actually collected.

In addition, I think, that if no time is stated with the data emonCMS (does what it should) and assumes local time, but I think that is **server** local not **user** local (but I could be wrong on this one).

Second addition, the same it true for the MQTT & full JSON data. This data can include a **time** element can be inserted with a timezone (which will be set by the sensor) or without, in which case the **server** local timezone would be used.

I seem to remember at one point having trouble with the data being an hour out and I think the server was on UTC TZ at the time, but I was on BST (+1).

chaveiro, 2020-04-19:
> At present both user and browser timezones need to be the same for many of the visualisations to work correctly. E.g if there is a mismatch, the current day will disappear off the my electric and my heatpump graphs produced using the delta mode..

Maybe the next step could be enunerate exactly what issues exist currently on the scenario "User TZ i Different" from browser TZ.

borpin, 2020-04-20:
https://community.openenergymonitor.org/t/having-trouble-with-dates-and-times-in-data-export-is-the-options-dialogue-in-bst-while-the-backend-in-utc/13914

Looks like this is using browser TZ?

TrystanLea, 2020-04-25:
What is the exact format that you would expect to see in the export @borpin I've been reading through https://en.wikipedia.org/wiki/ISO_8601 and https://stackoverflow.com/questions/5322285/how-do-i-convert-datetime-to-iso-8601-in-php looks like quite a few options.

Current format "d/m/Y H:i:s":

    25/04/2020 00:00:00

"d/m/Y H:i:s T" would give:

    25/04/2020 00:00:00 BST

"c" or DateTime::ATOM:

    2020-04-06T10:00:00+01:00

This is a really easy thing to change

chaveiro, 2020-04-25:
The exported data is in CSV to be used directly on Excel that only supports local datatime strings in CSV format. 
Adding iso8601 format break CSV in excel requiring further work with macros to parse it. 
Im sure end users dont want that.

borpin, 2020-04-25:
> Adding iso8601 format break CSV in excel requiring further work with macros to parse it.
Well, a simple function.

@TrystanLea the key thing is that without any TZ data, the string is treated as local time (ISO standard) and that should be made clear that is what is returned.

If I change the TZ field, the CSV data stays the same. The time string should match the selected TZ and the epoch numbers stay the same (but the date time string does not change). Actually, it currently does the right thing and only returns the data in local time 😀 

The key point is it should be clear what the date-time string is and what it relates to. You could add an additional dropdown for ISO format and I'd go for `2020-04-06T10:00:00+01:00` as I think that is what many languages use as their to ISO format.

chaveiro, 2020-04-25:
So Borpin, can you elaborate that text notice to the export window and submit a commit?

TrystanLea, 2020-04-25:
Thanks good to know @chaveiro re excel, so its not just emoncms then being ambiguous ;) a dropdown to select formats would be nice, the offset field is probably a bit confusing too.. the options should be: unix timestamp, excel compatible time and ISO format.. 

one for the long growing enhancement list...

TrystanLea, 2020-04-25:
I've had a quick look at adding a dropdown to the download dialog, it could look like this:

[image]

however the timezone offset is problematic.. I realise now that it's there as a work around for exporting daily data, daily data is not currently handled correctly by the csv_export function. Really the csv_export function needs rebuilding from scratch, to take into account a wider range of export requirements such as: timezones, providing averages when lower resolutions are selected (optional), aligning day, month etc to timezones.. 

You can already achieve a lot of this via the Graph interface - while it would be nice to have it available via the feeds page - there are I think other more pressing development priorities. 

The main difference with the approach taken by the csv_export function used for the dialog on the feed list is that it pipes the output into a download allowing large feeds to be downloaded in their entirety at full resolution, whilst the graph module uses the get_data and get_data_DMY functions, but these store the results in arrays in php before outputting as a json, which is then converted to csv in the graph module client.

Given that the current implementation of the csv downloader has its limitations, perhaps removing the offset, and daily, weekly, monthly and annual intervals is a good approach to remove confusion?
If its broken and you cant fix it immediately , perhaps its best to remove it?

TrystanLea, 2020-04-25:
Looking at the message given alongside this dialog, Im guessing mysql supports averaging as part of this @chaveiro? and perhaps better timezone support? 

PHPFina and PHPTimeSeries do not currently support averaging as part of csv_export nor do they handle the day/month/year timezones..

Another option here would be to pipe the output if the user selects a high resolution download but fallback to get_data and get_data_DMY where lower resolutions are required.. 

still lots of work involved.. at least Im a bit more aware of the problem here.. I guess I had personally stopped using this feature in favour of the graph module exporter which gives a more controlled result, though for a more limited number of datapoints in the export

borpin, 2020-04-25:
I think this (and the graph function that I was looking at), should make it clear you are selecting a local time, i.e. local to that screen. It can only use the browser time realistically.

The ISO format should simply be a translation of the epoch time using the php functions - nothing tricky I'd suggest. I do the reverse when taking the timestamp on the input processing.

chaveiro, 2020-04-25:
Mysql does averanging.
In my opinion there is no issue with the current export and ISO formats are not needed by anyone already using it regulary. - That are monstly business users for sure. 
Agree there are more important things to do.

borpin, 2020-04-26:
The one thing I have always struggled with is getting the actual datapoints from the time series store.

What I want to know is exactly what data is there and with what timestamp so I can look for reasons why data might not have been stored. It always seems hugely complex to do so.

TrystanLea, 2020-04-30:
I've made a start on improving the timezone support for the feed csv export tool, so far adding timezone support to daily, monthly and annual interval modes in PHPFina, I've also added in the dropdown selector above, this is work in progress and currently in a development branch see: https://github.com/emoncms/emoncms/compare/csv_export_phpfina_fix_dmy

The next step is to extend this to the other feed engines, to support interval codes d,w,m,y as indicators to output the result aligned to the timezone rather than a fixed interval

TrystanLea, 2020-04-30:
There's also a function to return the feed timezone, although it currently pulls that from the user table, at some point we can make the switch to storing timezones at the feed level. I think its a complementary function to your pull request here which Im still considering https://github.com/emoncms/emoncms/pull/1545/files @hutorny

TrystanLea, 2023-04-06:
Issue reopened here: https://github.com/emoncms/emoncms/issues/1823

closed 2023-04-06 by TrystanLea