emoncms/emoncms #1064: MQTT ID must be unique for Broker

borpin, 2018-10-12:
https://github.com/emoncms/emoncms/blob/22abb9523ee2c05943f9e8e6fdb54bc20ea2fa4f/scripts/phpmqtt_input.php#L105

In the unlikely (cough) situation where 2 instances of EmonCMS are connecting to the same broker, not having a uniqueID for the clientID will cause unexpected disconnections, which may result in several hours of head scratching.

It is suggested the ID is made unique by appending 4 digits of the API.

The original suggestion from @pb66 (below) of using the API key was not great as it could expose the key insecurely, but only using n number of chars would provide a unique ID.

```
$mqtt_client = new Mosquitto\Client($id=$userget_apikey_write($mqttsettings['userid'],$cleanSession=false));
```

Paul-Reed, 2018-10-12:
I've suggested previously, that [something like this would](https://github.com/knolleary/pubsubclient/commit/33170273a965c7ac3a93788bd1f2dbde3cbf8424) overcome the problem.

Paul

pb66, 2018-10-12:
> The original suggestion from @pb66 (below) of using the API key was not great as it could expose the key insecurely, but only using n number of chars would provide a unique ID.

I did say at the time it was not great and suggested we just break out the id to the mqtt settings which @TrystanLea agreed with. (https://github.com/emoncms/emoncms/pull/823#issuecomment-400455859)

The clientid will be for the whole emoncms instance, not for a particular user, so apikey (or part of) is not good, nor is mac address or ip as there could be more than one emoncms instance.

By having a (admin) user defined clientid in the settings.php it can be a familiar name for use with multiple emoncms instances on a secure network or LAN, or it can be a complex selection of chars to increase security, as the application dictates and it can be changed easily if compromised.

borpin, 2019-02-07:
#1160 has been merged which resolves this issue.

closed 2019-02-07 by borpin