emoncms/emoncms #1065: Redis and System-runner

borpin, 2018-10-13:
Following on from the work done on system-runner, once merged, do the install instructions for Redis need to refer to the installation of the new system-runner script & service?

https://community.openenergymonitor.org/t/rpi-3b-getting-hot-redis-high-cpu-usage-emonsd-13jun18-dev-specific/8263/23

TrystanLea, 2019-07-11:
Thanks @borpin do you know which install instructions that you had in mind need updating?

borpin, 2019-07-11:
Any other module that references or uses service-runner.  They should simply point to [this](https://github.com/emoncms/emoncms/blob/master/scripts/services/install-service-runner-update.md)

So 
* backup (no mention in readme and has it's own version of service-runner)
* sync (incorrect instructions in ReadMe)
* config (?)
* wifi

and there is a version of service-runner here https://github.com/openenergymonitor/emonpi/blob/master/service-runner

TrystanLea, 2019-07-12:
Great.

- I've updated the backup module readme and removed the local copy: https://github.com/emoncms/backup 
- Updated the sync module readme: https://github.com/emoncms/sync
- Added a note to the WIFI module readme https://github.com/emoncms/wifi
- config module does not use service-runner

service-runner in emonpi repo is needed for backwards compatibility I think.

borpin, 2019-07-12:
* config module

https://github.com/emoncms/config/blob/f5de426f52d793ff417fdbe585ca9a85d9818e76/config_controller.php#L171

(I just searched the organisation for 'service-runner' 😄)

closed 2023-02-10 by borpin