emoncms/emoncms #1069: Use of user "pi" in service-runner

pb66, 2018-10-16:
A good point was raised in the ["Added systemd based service-runner in python"](https://github.com/emoncms/emoncms/pull/1025#issuecomment-429472440) PR shortly before it was merged, I've opened this issue so the point isn't lost with the closed PR thread.

Use of user "pi" in the service unit was carried over from the old init script and it is specific to using a raspberry pi with the default user only.

The generic service-runner should not really be run as user "pi".

greeebs, 2018-10-18:
IMHO running it as root by default is also the wrong answer. Remember that this script takes whatever is on the unauthenticated "service-runner" redis key and then _runs it_. 

Hmm, I just checked the update script in the emonpi repo (which is what started the need for service-runner in the first place):
https://github.com/openenergymonitor/emonpi/blob/master/service-runner-update.sh
sudo is all over the place in that script so maybe root is the right answer for an emonpi?

What about other self installs though?

What are the implications on EmonCMS if a specific user is created for service-runner? What permissions are actually required for it to do its thing - are they just file permissions? If so, should we create a emonAdmin group and make sure all of the relevant files and folders are `chgrp emonAdmin` and `chmod g+w`, then add the service-runner user to that group?

TrystanLea, 2019-07-11:
Noting this to review as part of generic target OS installation compatibility, to come back to after emonSD build script release.

borpin, 2019-07-11:
Could you add a label for those Issues you have 'Noted' please @TrystanLea?

TrystanLea, 2023-04-06:
Pretty sure this issue was fixed, the user can be set using:

    sudo systemctl edit service-runner.service

See:https://github.com/emoncms/emoncms/blob/master/scripts/services/install-service-runner-update.md

closed 2023-04-06 by TrystanLea