emoncms/emoncms #1074: Input Process Stopped working after EmonPi Update

Paulf007, 2018-10-20:
System Info:
<details><summary>Server Information</summary><pre>

| | | |
| --- | --- | --- |
|Emoncms|Version|low-write 9.9.0
||Modules|Administration : App v1.2.0 : Backup v1.1.5 : EmonHub Config v1.0.0 : Dashboard v1.3.1 : Device v1.1.1 : EventProcesses : Feed : Graph v1.2.1 : Input : Postprocess v1.0.0 : CoreProcess : Schedule : Network Setup v1.0.0 : sync : Time : User : Visualisation : WiFi v1.3.0
||Git URL|https://github.com/emoncms/emoncms.git
||Git Branch|* stable
||Buffer|<span id="bufferused">loading...</span>
||Writer|Daemon is running with sleep 60s
|Server|OS|Linux 4.9.35-v7+
||Host|emonpi emonpi (192.168.8.114)
||Date|2018-10-20 07:40:22 SAST
||Uptime| 07:40:22 up 16 min,  0 users,  load average: 0.93, 0.78, 0.51
|HTTP|Server|Apache/2.4.10 (Raspbian) HTTP/1.1 CGI/1.1 80
|MySQL|Version|5.5.57-0+deb8u1
||Host|127.0.0.1 (127.0.0.1)
||Date|2018-10-20 07:40:22 (UTC 02:00‌​)
||Stats|Uptime: 970  Threads: 5  Questions: 34042  Slow queries: 0  Opens: 93  Flush tables: 1  Open tables: 86  Queries per second avg: 35.094
|Redis|Version|2.8.17
||Host|localhost:6379 (127.0.0.1)
||Size|<span id="redisused">145 keys  (509.54K)</span>
||Uptime|0 days
|MQTT|Version|1.4.14
||Host|localhost:1883 (127.0.0.1)
|Pi|Model|Raspberry Pi 3 Model B Rev 1.2 - 1 GB (Sony UK)
||SoC|Broadcom BCM2835
||Serial num.|C5891ABA
||Temperature|CPU: 49.39°C - GPU: 48.3'C
||Release|emonSD-26Oct17
||File-system|Current: read-only - Set root file-system temporarily to read-write, (default read-only) 
|Memory|RAM|Used: 33.25% Total: 970.93 MB Used: 322.82 MB Free: 648.11 MB
|Disk|Mount|Stats
||/|Used: 67.07% Total: 3.33 GB Used: 2.23 GB Free: 957.8 MB
||/boot|Used: 36.36% Total: 59.95 MB Used: 21.8 MB Free: 38.15 MB
||/home/pi/data|Used: 13.06% Total: 968.33 MB Used: 126.5 MB Free: 791.84 MB
|PHP|Version|5.6.37-0+deb8u1 (Zend Version 2.6.0)
||Modules|apache2handler : bcmath : bz2 : calendar : Core v5.6.37-0+deb8u1 : ctype : curl : date v5.6.37-0+deb8u1 : dba : dio v0.0.4RC4 : dom v20031129 : ereg : exif v1.4 $Id: cad29b729548e4206f0697710cc9e177f26fdff3 $ : fileinfo v1.0.5 : filter v0.11.0 : ftp : gd : gettext : hash v1.0 : iconv : json v1.3.6 : libxml : mbstring : mcrypt : mhash : mosquitto v0.3.0 : mysql v1.0 : mysqli v0.1 : openssl : pcre : PDO v1.0.4dev : pdo_mysql v1.0.2 : pdo_pgsql v1.0.2 : pdo_sqlite v1.0.1 : pgsql : Phar v2.0.2 : posix : readline v5.6.37-0+deb8u1 : redis v2.2.7 : Reflection v$Id: 5f15287237d5f78d75b19c26915aa7bd83dee8b8 $ : session : shmop : SimpleXML v0.1 : soap : sockets : SPL v0.2 : sqlite3 v0.7-dev : standard v5.6.37-0+deb8u1 : sysvmsg : sysvsem : sysvshm : tokenizer v0.1 : wddx : xml : xmlreader v0.1 : xmlwriter v0.1 : Zend OPcache v7.0.6-devFE : zip v1.12.5 : zlib v2.0
</pre></details>
<details><summary>Client Information</summary><pre>

| | | |
| --- | --- | --- |
|HTTP|Browser|Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36
|Screen|Resolution|1680 x 1050
|Window|Size|1848 x 954
</pre></details>

Issue: 
I did upgrade via the Admin console of EmonPi. After the update the "Input Process" stopped working. 
The data that is sent to emon is comming through (Via node red and the Emon Node) but it is no longer been processed. 
See screen shots 
[image]
[image]
[image]
 I can also not add an process as it seems missing.
Looking at the console on the admin page if I attempt to upgrade emonpi again I get the following

Starting emonPi Update >
via service-runner-update.sh
Service Runner update script V1.1.1
EUID: 1000
Argument: emonpi
Sat 20 Oct 07:46:12 SAST 2018
#############################################################

emonSD version: emonSD-26Oct17

ERROR: emonSD base image old or undefined...update will not continue
See latest verson: https://github.com/openenergymonitor/emonpi/wiki/emonSD-pre-built-SD-card-Download-&-Change-Log
Stopping update

Is there a way to fix this?

pb66, 2018-10-20:
Apparently similar issue on forum

https://community.openenergymonitor.org/t/after-update-to-9-9-0-process-list-all-unavaiblable/8893?u=pb66

pb66, 2018-10-20:
@Paulf007, the content of your issue isn't displaying correctly, but I did find and read your comments, I believe someone has reported a similar issue on the OEM forum, (as yet there has been no reply).

I would always recommend posting on the OEM forum in the first instance as assistance tends to be more readily available there and in this particular instance, your full post will display correctly there. 

If you go to edit mode for you post above, you should be able to copy the whole post and paste it into a post on the OEM forum without any changes or rewriting.

(I have also an issue about the formatting issue #1076)

glynhudson, 2018-10-21:
Fixed by https://github.com/emoncms/emoncms/pull/1077

closed 2018-10-21 by glynhudson