emoncms/emoncms #1078: Horizontal scrolling not working work on mobile devices

glynhudson, 2018-10-23:
Running latest 9.9.2 horizontal scrolling doesn't work on mobile devices, this is particularly an issue when device module beta is switched off. 

The issue only effects the Inputs and Feeds page, I can horizontal scroll on the admin page. Could this be a swype gesture issue? 

[image]

Paul-Reed, 2018-10-23:
Hhmm, I didn't think that the feeds page should render like that on a mobile device??
Normally CSS hides some of the columns so that it fits.

emrysr, 2018-10-23:
hi @glynhudson, I can't re-produce this... The feed and input view fit resize the list to fit my screen so horizontal scrolling is not required.

I have now removed the scrolling interaction js (hammer.js) if the beta features are not switched on.

could you let me know if this is still an issue on your mobile?

TrystanLea, 2018-10-24:
Thanks for @emrysr's fix solving the scrolling issue.

The new inputs and feeds interface is designed to solve the long running issue with mobile support.

closed 2018-10-24 by TrystanLea