emoncms/emoncms #1081: Unable to set up any feeds/processes

petrklus, 2018-10-23:
Hi,

I've received a fully-assembled emonPI and when it booted up, I've performed both updates (one after another).

After rebooting, I cannot seem to be able to add any feeds or set up inputs:

[image]

Even if I somehow manage to sift through the "undefined" entries, I then get this when trying to save:

[image]

Any idea what may be going wrong?

Server info below:
```<details><summary>Server Information</summary><pre>

| | | |
| --- | --- | --- |
|Emoncms|Version|low-write 9.9.2
||Modules|Administration : App v1.2.0 : Backup v1.1.5 : EmonHub Config v1.0.0 : Dashboard v1.3.1 : Device v1.1.1 : EventProcesses : Feed : Graph v1.2.1 : Input : Postprocess v1.0.0 : CoreProcess : Schedule : Network Setup v1.0.0 : sync : Time : User : Visualisation : WiFi v1.3.0
||Git URL|https://github.com/emoncms/emoncms.git
||Git Branch|* stable
||Buffer|<span id="bufferused">loading...</span>
||Writer|Daemon is running with sleep 60s
|Server|OS|Linux 4.9.35-v7+
||Host|emonpi emonpi (192.168.2.18)
||Date|2018-10-23 19:29:08 UTC
||Uptime| 19:29:08 up 28 min,  0 users,  load average: 1.54, 1.55, 1.32
|HTTP|Server|Apache/2.4.10 (Raspbian) HTTP/1.1 CGI/1.1 80
|MySQL|Version|5.5.57-0+deb8u1
||Host|127.0.0.1 (127.0.0.1)
||Date|2018-10-23 19:29:08 (UTC 00:00‌​)
||Stats|Uptime: 1684  Threads: 3  Questions: 893  Slow queries: 0  Opens: 61  Flush tables: 1  Open tables: 54  Queries per second avg: 0.530
|Redis|Version|2.8.17
||Host|localhost:6379 (127.0.0.1)
||Size|<span id="redisused">55 keys  (557.55K)</span>
||Uptime|0 days
|MQTT|Version|1.4.14
||Host|localhost:1883 (127.0.0.1)
|Pi|Model|Raspberry Pi 3 Model B Rev 1.2 - 1 GB (Stadium)
||SoC|Broadcom BCM2835
||Serial num.|5468D824
||Temperature|CPU: 61.22°C - GPU: 61.2'C
||Release|emonSD-26Oct17
||File-system|Current: read-only - Set root file-system temporarily to read-write, (default read-only) 
|Memory|RAM|Used: 26.40% Total: 970.93 MB Used: 256.35 MB Free: 714.58 MB
|Disk|Mount|Stats
||/|Used: 60.97% Total: 3.33 GB Used: 2.03 GB Free: 1.14 GB
||/boot|Used: 36.32% Total: 59.95 MB Used: 21.77 MB Free: 38.17 MB
||/home/pi/data|Used: 1.09% Total: 3.46 GB Used: 38.77 MB Free: 3.25 GB
|PHP|Version|5.6.30-0+deb8u1 (Zend Version 2.6.0)
||Modules|apache2handler : bcmath : bz2 : calendar : Core v5.6.30-0+deb8u1 : ctype : curl : date v5.6.30-0+deb8u1 : dba : dio v0.0.4RC4 : dom v20031129 : ereg : exif v1.4 $Id: 1c8772f76be691b7b3f77ca31eb788a2abbcefe5 $ : fileinfo v1.0.5 : filter v0.11.0 : ftp : gettext : hash v1.0 : iconv : json v1.3.6 : libxml : mbstring : mcrypt : mhash : mosquitto v0.3.0 : mysql v1.0 : mysqli v0.1 : openssl : pcre : PDO v1.0.4dev : pdo_mysql v1.0.2 : Phar v2.0.2 : posix : readline v5.6.30-0+deb8u1 : redis v2.2.7 : Reflection v$Id: 5f15287237d5f78d75b19c26915aa7bd83dee8b8 $ : session : shmop : SimpleXML v0.1 : soap : sockets : SPL v0.2 : standard v5.6.30-0+deb8u1 : sysvmsg : sysvsem : sysvshm : tokenizer v0.1 : wddx : xml : xmlreader v0.1 : xmlwriter v0.1 : Zend OPcache v7.0.6-devFE : zip v1.12.5 : zlib v2.0
</pre></details>
<details><summary>Client Information</summary><pre>

| | | |
| --- | --- | --- |
|HTTP|Browser|Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36
|Screen|Resolution|1680 x 1050
|Window|Size|1665 x 899
</pre></details>```

glynhudson, 2018-10-24:
Hi, sorry to hear your having issues. Could you attach a screen grab of your Inputs list?

Can you log a single Input to Feed without any `+inp` input processing?

TrystanLea, 2018-10-24:
Hello @petrklus, looks like the same issue as https://community.openenergymonitor.org/t/after-update-to-9-9-0-process-list-all-unavaiblable/8893/4

This is caused by browser caching of old javascript and can be fixed by clearing browser cache. Emoncms should with update 9.9.1 do this automatically but its not universally supported across browsers.

Could you try clearing your browsers cache?

perefeliu, 2018-12-02:
Thanks! I used instead the browser in private navigation and ti worked for me!

closed 2018-12-10 by glynhudson