emoncms/emoncms #1089: Present working directory not given when service ran from service-runner

TrystanLea, 2018-10-26:
Hello @greeebs, testing the new python service-runner here. Im having trouble running the sync and postprocess scripts as the present working directory is not returned. Works fine from command line but not from the service-runner. The first line here returns / instead of /home/pi/sync
 
   DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
    sudo php $DIR/sync_run.php

Do you have any ideas?

greeebs, 2018-10-27:
I'll have a look tonight - I have a few ideas on why.

greeebs, 2018-10-27:
By default, the python call being used to execute the program (`subprocess.call()`) uses `/bin/sh`. This may or may not actually be `bash` depending on the distribution you're using. If you replace `${BASH_SOURCE[0]}` with `$0` it should work in both `sh` and `bash`:
```
DIR="$( cd "$( dirname $0 )" && pwd )"
```
Whats the shebang (#!) at the top of your script file that contains those lines? Is there one?

greeebs, 2018-10-27:
I found this file: https://github.com/emoncms/postprocess/blob/emonpi/postprocess.sh, which is what I assume you're talking about. It will probably work perfectly fine if you just add this to the first line:
```
#!/bin/bash
```

greeebs, 2018-10-27:
Adding the shebang definitely should fix this.
Here's the same script run from `bash`, then from `sh`:
```
pi@emonpi(rw):~$ ./a.sh
/home/pi
pi@emonpi(rw):~$ sh
$ ./a.sh
/home/pi
$ exit
pi@emonpi(rw):~$ cat a.sh
#!/bin/bash
DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
echo $DIR
```

glynhudson, 2018-10-27:
ah thanks a lot, dam should have noticed that! Issue has now been fixed with shebang https://github.com/emoncms/sync/commit/30bba0a0c00d190828010ba65d4f6d60ff24566d

closed 2018-10-27 by glynhudson