emoncms/emoncms #1093: inpu/post: data items may not contain / character

karlp, 2018-10-28:
Sending data like 
```
https://emon.example.org/input/post?node=emontx&fulljson={"power/1":100,"power/2":200,"power/3":300}
```

will be accepted, and the input list will show the point names "cleaned" to "power1", but every new post will create a new point with the same name, and the inputs table ends up with (hundreds) of duplicates.

I don't mind the automatic fixup particularly, (makes a few apis a little easier to not have to worry about tthe / I guess) but if you're going to fixthem dynamically, it at least needs to be complete.  Alternatively, simply rejecting the post would be fine.

found with version: v9.5.1-1500-gbaa6d2f

borpin, 2018-11-02:
> will be accepted, and the input list will show the point names "cleaned" to "power1", but every new post will create a new point with the same name, and the inputs table ends up with (hundreds) of duplicates.

I have just found the same with MQTT topics and colons #1105 

@karlp How did you clean out the additional inputs?

karlp, 2018-11-02:
@borpin mysql at the command line and just dropping them from the input table.  that git tag was at most two days old when I filed this.  That's the output of "git describe" so it's 1500 commits past the v9.5.1 tag, at git commit baa6d2f

(I was also feeding mqtt topics directly in, I hadn't yet found the documentation on using the mqtt apis for feeding data)

borpin, 2018-11-02:
@karlp 

> mysql at the command line and just dropping them from the input table

I thought so but have not a clue how to!! Could you provide some more detailed help via the [community](https://community.openenergymonitor.org/t/delete-multiple-input-nodes-100s/9034)?

> That's the output of "git describe" so it's 1500 commits past the v9.5.1 tag, at git commit baa6d2f

Ah OK. Interesting, wonder why it is picking it up from such an old tag. Git is a weird and wonderful tool 😄. Just did the same and got (almost) the same result.

glynhudson, 2018-11-03:
>That's the output of "git describe" so it's 1500 commits past the v9.5.1 tag, at git commit baa6d2f

>Ah OK. Interesting, wonder why it is picking it up from such an old tag. Git is a weird and wonderful tool 😄. Just did the same and got (almost) the same result.

I think this is because our release tags are created via github.com so they are so-called 'lightweight' tags using `git describe --tags` will return the correct tag `git describe --tags --debug` will show what's going on

```
git describe --tags 
V9.9.2-51-g1e8a85b4
```

----
```
git describe --tags --debug
searching to describe HEAD
 lightweight       51 V9.9.2
 lightweight       54 v9.9.1
 lightweight       59 v9.9.0
 lightweight      543 9.8.31
 lightweight      625 9.8.30
 lightweight      630 9.8.29
 lightweight      742 9.8.28
 lightweight      780 9.8.27
 lightweight      794 9.8.25
 lightweight      799 9.8.24
traversed 813 commits
more than 10 tags found; listed 10 most recent
gave up search at 2e27b237c2514222b6909e4a5d901aaa795b1f5a
V9.9.2-51-g1e8a85b4
```

borpin, 2018-11-03:
Hi @TrystanLea, thanks - every day a learning day. 😄

TrystanLea, 2019-07-11:
Is this issue ok to close?

borpin, 2019-07-11:
No I don't think so.  My fix #1156 just fixed the colons. I did not know what other characters were regarded as illegal in an input name (such as '/') so I suggest a more robust means of cleaning them out!

https://github.com/emoncms/emoncms/blob/e08ee816b79ece03b51d0b1e2d7f0350034c8d3f/scripts/phpmqtt_input.php#L233-L233

TrystanLea, 2019-07-12:
Thanks @borpin I've tested with @karlp's example above and can confirm this issue is still present.

TrystanLea, 2019-07-12:
Implemented a fix for the http api: https://github.com/emoncms/emoncms/commit/fc77f92e554b01ab7c1bbcccd9829adfb46c857d I will add a similar check to emoncms_mqtt

TrystanLea, 2019-07-12:
I've created a similar change to the emoncms_mqtt script in a seperate branch this time https://github.com/emoncms/emoncms/tree/filter_name_nodeid - I should have included both together in this branch for testing.

borpin, 2019-07-12:
On the mqtt side, can you take out my ':' fix (linked to above) then? That was just quick and dirty to fix my immediate issue (assuming that your fix fixes that!).

TrystanLea, 2020-01-20:
filter_name_nodeid was merged

closed 2020-01-20 by TrystanLea