emoncms/emoncms #1104: Version numbers and tags

borpin, 2018-11-02:
Hi, can the version numbering please be consistent.

Firstly the GitHub 'tags' are case sensitive so 9.9.3 is different to v9.9.3 and V9.9.3.  Lowercase has been used most so I suggest we stick with that.

Secondly I suggested a 'beta' in the version number.  I note that there was a 9.9.2-beta that became 9.9.2 - Great.

When that was published (i.e. a tag created) the first change to master should be to bump the version to 9.9.3-beta (rather than 9.9.3 IMHO).

In theory we could just leave it as 9.9.3 and just bump the version each time a tag is created, but the 'beta' does act as a warning I feel.

glynhudson, 2018-11-02:
Totally agree, I've jump bump the current master to beta. 

Apologies it's my fault I messed up the last couple of release tags using upper than lower case 'V'. The emoncms convention has always been to not have any 'v' in the release tag. I'm sure if it's possible to change the tag retrospectively? Ensure no 'v' is present in the future.

closed 2018-11-02 by glynhudson

borpin, 2018-11-02:
No bother. The variation on the tags have appeared over time.  I'd be inclined to stick with what the last few tags have been as they are at the top!

TrystanLea, 2018-11-03:
I've just merged master to stable, I removed the beta flag, then merged. 

Do I understand correctly that on the next commit to master we switch the version to 9.9.5-beta?

glynhudson, 2018-11-03:
>Do I understand correctly that on the next commit to master we switch the version to 9.9.5-beta?

Correct. This will then become 9.9.5 when we do the next stable release (drop the beta).

Also remember to make a release with change notes for a stable release: https://github.com/emoncms/emoncms/releases

>The variation on the tags have appeared over time. I'd be inclined to stick with what the last few tags have been as they are at the top!

Mmm tricky, one the precedence of having no 'v' has been going on much longer than the last two releases with a 'v'. 

Of course, blame myself for this, but github has a role to play, it should be possible to define a release style template for a project. Doing a release is complicated enough already. Making a release tag with release notes should quick and simple.

borpin, 2018-11-03:
Hi @TrystanLea I suggest that if you do a merge, then version should go from `9.9.x-beta` to `9.9.x` - merge - `9.9.(x+1)-beta`

That would be my preferred process.

BTW it has been pointed out by @karlp (#1093) that `git describe` references back to v9.5.1 `v9.5.1-1500-gbaa6d2f`.  Is this a function of how you are publishing versions?

glynhudson, 2018-11-03:
>BTW it has been pointed out by @karlp (#1093) that git describe references back to v9.5.1 v9.5.1-1500-gbaa6d2f. Is this a function of how you are publishing versions?

I think this is because our release tags are created via github.com so they are so-called 'lightweight' tags using `git describe --tags` will return the correct tag `git describe --tags --debug` will show what's going on

commit 6d244d77 (2018-11-02): bump beta version fix #1104