emoncms/emoncms #1105: MQTT input does not handle ':' (colon) in topics

borpin, 2018-11-02:
A topic of `emon/xiaomi/4c:65:a8:d9:59:1a/battery` results in a new feed being created with every data item via MQTT.

borpin, 2019-01-17:
Merge of #1156 solves the specific case of colons, but a check is required to see if there are other characters that could cause issues.

TrystanLea, 2020-03-20:
Here's the result with your example above and the latest topic mapping: 

[image]

Then with: 

    emon/xiaomi/4c,65*a8/d9^59;1a/battery

[image]

borpin, 2020-03-20:
Ah Ok great. Cheers.

closed 2020-03-20 by borpin