emoncms/emoncms #1107: MQTT topics accepted when invalid

borpin, 2018-11-02:
I presume this code was intended to check to see if the MQTT topic was too deep.

It does not work ☹️ (as I discovered accidentally).

Instead, could you simply do a split and count the size of the 2 arrays ($topic and $basetopic) and if difference is greater than 2 reject the message?

https://github.com/emoncms/emoncms/blob/4e9f8d8c01852962e29d1ccfcde24278e21fa996/scripts/phpmqtt_input.php#L272-L287

TrystanLea, 2019-01-10:
Thanks @borpin, will look into it

borpin, 2019-01-10:
I did a quick and dirty fix in #1106

TrystanLea, 2020-03-05:
I recently found myself affected by this issue when trying to subscribe to data posted by the OVMS vehicle monitor. It includes topics such as:

    emon/ovms/metric/v/b/soc
    emon/ovms/metric/v/b/c/voltage/dev/max

This would of course not work with the current topic mapping implementation. I think a good mapping approach would be to map:

    emon/ovms/metric/v/b/soc:

to be:

    nodeid: ovms
    input name: metric_v_b_soc

and:

    emon/ovms/metric/v/b/c/voltage/dev/max

to be:

    nodeid: ovms
    input name: metric_v_b_c_voltage_dev_max

I've got an implementation that provides this mapping here:
https://github.com/emoncms/emoncms/compare/emoncms_mqtt_topic_filter

its backwards compatible with the standard topic format e.g emon/emontx/power1

What do you think @borpin @chaveiro

TrystanLea, 2020-03-19:
I've merged this in, closing for now.

closed 2020-03-19 by TrystanLea