emoncms/emoncms #1115: Align output of 'git describe' with released version number

borpin, 2018-11-14:
Following on from #1104 and related to  #1114....

@glynhudson - following on from our discussion on tags, I have done a few experiments (I am not a git guru) and think there is a simple solution so `git describe` describes the latest version on stable correctly.

Once all the commits have been merged into the stable branch (including the updated `version.txt` - say v9.9.5), use the following commands (I don't think the checked out branch matters as it is effectively putting a marker down relative to the last commit):

```
git tag v9.9.5 -m "short message here"
git push origin v9.9.5
```
It is now possible to create a release from Github using the `v9.9.5` tag.

Following a `git pull` a `git describe` on stable branch should give `v9.9.5` and from master `v9.9.5-xx-yyyyyyyy` where xx is the number of commits since the tag was created and yyyyyyy is the last commit number.

By including the output of `git describe` in the administration screen, it will be immediately clear what version the user is running.

I would appreciate someone with greater knowledge of `git` checking this out, but it would be nice and clean.

glynhudson, 2018-11-15:
That's good to know. I'll try it out next time we do a release. However, I personally think the easiest and most reliable method of checking what version your running is to look at the footer of the webpage or via shell at the contents of `version.txt`. 

To check the version of Modules looks at the contents of `module.json`, the version number from `module.json` is displayed on the admin page.

closed 2018-11-15 by glynhudson

borpin, 2018-11-16:
Except of course, it is not *reliable* when on master as you have no idea when it was last pulled.  Including an annotated tag would then give you that information.