emoncms/emoncms #1116: Smoothie chart: No data from feed api

Tim-Otte, 2018-11-16:
When using a smoothie diagram, no data from the feed is displayed. This is because the GET parameter _mode_ is missing in the `smoothie.php` in the _vis_feed_data_ function of the AJAX call. The PHPFina engine always returns `false` if this parameter is not set to `"daily"`, `"weekly"` or `"monthly"`. If you insert this parameter manually, an empty array will be returned.

Please check this error and give me a short feedback.

TrystanLea, 2018-11-16:
Hello @Tim-Otte, provisional fix: https://github.com/emoncms/emoncms/commit/d471c07deff8c2a8472f34d0f80001ae910d99ab

Have you tried the realtime data visualisation too? It gives more detail

Tim-Otte, 2018-11-20:
Thanks for the provisional fix. Now it works as a charme.

I've tried it with both, but none worked worked for me...

closed 2018-11-20 by Tim-Otte