emoncms/emoncms #1128: min max values for a certain time frame?

daniu-de, 2018-11-21:
Hi

not sure this is the right place for a feature request but I think it would be cool to have some way to get min/max values for a certain timeframe (not sure what would be best: last 24h or a certain date or defineable or whatever

regards

borpin, 2018-11-21:
Assuming you have installed the 'graph' module, if you click on a feed name (from the feeds page), under the graph that is displayed, is **Show Statistics** - it not very clear (@TrystanLea - it's not just me 😄)

[image]

Click on that and you get...

[image]

daniu-de, 2018-11-21:
well ... that is not exactly what I want. What I miss is an ability to add the min/max value to a dashboard.

I got a dashboard with a table with all current values of certain sensors. It would be cool to be able to add a min/max value for - let's say the last 24h or the like (a bit like the weather foarcast)

borpin, 2018-11-21:
Ah OK, you did not mention a dashboard 😄 

The Max/Min values are calculated by the javascript on the page from the data returned from an API call, so I am sure it is possible to generate a custom widget to do what you want, but that is way outside my skill set I am afraid!  You might get more response on the community site.

Paul-Reed, 2018-11-21:
If you look at your input processes, you'll see that there is an option to select `Max daily value` and `Min daily value` and save that to a feed.
That gets reset at midnight, and you can use that feed to either display on a dashboard 'feedvalue' widget, or use it as an additional setting in one of the widgets - such as the thermometer widget (screenshot below).

Is this the sort of thing you wanted?

[image]

[image]

closed 2019-07-11 by TrystanLea