emoncms/emoncms #1129: text on the administration tab that says “Set root file-system temporarily to read-write, (default read-only)”.

TrystanLea, 2018-11-21:
https://community.openenergymonitor.org/t/bug-read-write-set-root-file-system-temporarily-to-read-write-default-read-only/9235

TrystanLea, 2019-01-10:
Did we discuss a solution for this one @glynhudson @emrysr, what is the best way to detect the image version and change this entry as applicable?

emrysr, 2019-01-23:
i'd assume that the command `rpi-rw` and `rpi-ro` commands are now not available on the new image?

we could test for that?

pb66, 2019-01-23:
I thought that was the best way too. Only show the button if the command or calls exists.
https://community.openenergymonitor.org/t/bug-read-write-set-root-file-system-temporarily-to-read-write-default-read-only/9235/5?u=pb66

PR #1179 (2019-01-29): checked for rpi-rw script to identify older version
fix #1129 

this line will now be hidden (`/emonmcms/admin/view`) on the newer images of emoncms that don't have the filesystem read only restriction:
[image]

```
if (file_exists('/usr/bin/rpi-rw')) {
```
closed 2019-01-29 by TrystanLea