emoncms/emoncms #1132: Feature request - Graph module - Scale both Y Axis

borpin, 2018-11-23:
Offer the ability to scale both Y Axis independently when 2 or more sources, using L & R axis are selected.

[image]

Also offer a 'reset' option back to auto (what does reload actually do?).

LeaseOnLife, 2019-01-10:
yes please, is there a way to up-vote this?

djh42, 2019-03-05:
I would like to see this enhancement too.
> Also offer a 'reset' option back to auto
That would be useful - it seems strange to actually have to type 'auto' in the correct case as well :)
> (what does reload actually do?).
Is that not self-evident? Surely, it reloads the data and displays it according to the current settings?
It's not clear to me why some changes display immediately, whilst others require a reload to see the effect.

closed 2019-08-05 by TrystanLea