emoncms/emoncms #1133: mouse-over to read value of graph does not work when graph is used in dashboard

LeaseOnLife, 2018-11-28:
I believe this worked a few releases back. Tested with Firefox and Chrome on Linux. 
Mouse-over works fine in graphs using Setup/Graphs. But when that same graph is used in a dashboard, the mouse-over does not show values anymore. It does show the circle to be read out on the graph, but the pop-up containing value name and value does not come up.

Can anyone confirm?
Dirk

TrystanLea, 2018-12-19:
Thanks Dirk, this has been highlighted before but Im struggling to find the posts. Testing here with v.9.9.6-beta and latest graph module it seems to be working ok, here's a screenshot of a graph embeded in a dashboard:

[image]

LeaseOnLife, 2019-02-20:
whack-a-mole ? 
I just updated emoncms, not using git but downloaded 9.9.6 tgz 

I then added the modules dashboard and graph from git. I am back at not seeing the mouse-over values. Anything I am doing wrong?
Dirk

LeaseOnLife, 2019-02-28:
fixed in 9.9.7, thanks!

closed 2019-02-28 by LeaseOnLife