emoncms/emoncms #1137: incorrect data recorded in feeds

djh42, 2018-12-11:
In my system I see incorrect data values recorded for power_to_kwh and wh_accumulator type feeds. Specifically they both include retrograde data, which ought to be impossible for those feed types.

See https://community.openenergymonitor.org/t/getting-started-with-emoncms/9356/20 onwards for details. I can attach copies of images here if it's worth while.

djh42, 2018-12-14:
FWIW, there's even an example of a negative spike shown in the documentation!

See the graph under the text 'Raw data wh accumulator view:' in the page https://guide.openenergymonitor.org/setup/daily-kwh/

Oh, and I now realize that power_to_kwh can have retrograde data, if the underlying power is negative (e.g. when solar is exporting), but (a) that isn't the case for the data sets in my example, and (b) it isn't the case for wh_accumulator on a revenue meter at all.

TrystanLea, 2018-12-19:
Fix as discussed in testing here: https://github.com/emoncms/emoncms/tree/fix_padding_spikes

TrystanLea, 2019-01-10:
merged into master

closed 2019-01-10 by TrystanLea