emoncms/emoncms #1150: Background colours for service status list on admin page

TrystanLea, 2019-01-10:
I've merged pull request: https://github.com/emoncms/emoncms/pull/1134

@glynhudson @borpin @Paul-Reed the services all show up as red if not running, which is fine for emonpi/emonbase but may be confusing for installs that dont use these services as they are all optional.

I could remove the background colouring if $allow_emonpi_admin is true perhaps? do you have any preferences?

I removed the start,stop,restart and kill buttons for now as they are not yet implemented and I have not yet added any access to logs, which would be great to add in future.

borpin, 2019-01-10:
You could always check the status of the service?  Can't check how from here, but would it not be better, rather than just look for a process (which is a very binary state), to use `systemctl` to understand whether it has failed, not enabled etc?  I note that `systemctl status` command implicitly attempts to load any service not loaded so it may be better to do a `systemctl is-enabled` or a `systemctl show` instead (can't check if the output is usable right now).

borpin, 2019-01-10:
So a quick test here;
```
~# systemctl show service-runner.service | grep State

LoadState=loaded
ActiveState=active
SubState=running
UnitFileState=enabled
StateChangeTimestamp=Tue 2018-11-20 20:24:50 GMT
StateChangeTimestampMonotonic=6115500
```
On a non existent service
```
LoadState=not-found
ActiveState=inactive
SubState=dead
StateChangeTimestampMonotonic=0
```
I think this would be better than a `ps` command to determine state.

glynhudson, 2019-01-10:
Nice update. Yes, probably best remove for non emonPi systems to avoid confusion.

borpin, 2019-01-10:
No, I think you could check for all systems and just include when it is found.  I'd certainly like it.  The command is non-intrusive so checking when it does not exist would not be an issue.

Not sure how it would respond if it was still the init.d version though.

borpin, 2019-01-14:
@glynhudson @emrysr I have been trying to load the output of the command `systemctl show service-runner.service | grep State` into an array and then put the output into the admin view without success - my PHP is really not up to it. The only way I've found was to call the command multiple times on one service which is not very efficient.

If I could, I'd do a PR 😦!

I'd suggest putting the output into the view as 

* UnitFileState (but need to check for a null as if the service does not exist, command will return nothing)
* LoadState
* ActiveState
* SubState

Cheers - a really good enhancement.

PR #1153 (2019-01-16): hidden uninstalled services
showing green if  installed and running
showing red if installed and not running

fix #1150 

[image]

thanks @borpin for the `systemctl` command example in issue #1150
PR #1157 (2019-01-16): Feature/issue 1150 service statuses
fix #1150 

Added to this pr #1153 

Thanks @borpin for the suggestions regarding outputting usefull service statuses passed from:
```
systemctl show emonhub.service | grep State
```

Full service name now shown in list:
[image]
closed 2019-01-16 by TrystanLea