emoncms/emoncms #1162: Modules list in Admin

chaveiro, 2019-01-17:
It's difficult to maintain EmonCms and its modules updated.
A nice feature to add is a list on the admin page with:
 - list of available modules available for installation
 - current installed module version
 - newer available versions 
 - button to update / remove / install from git automatically

To whom is familiar with Arduino IDE plugins manager, it's the same feature.

borpin, 2019-01-17:
Agreed, but I suspect the starting point is to make all modules the same in terms of structure (master and stable) and version file.

There was some discussion around this when the version numbers were introduced into the admin page as some 'Modules' are core and some are optional.

chaveiro, 2019-01-17:
Or maintain a file with links to the 'stable' releases of the available official modules on the core repo and give the option for the user to enter the module url manually (similar to arduino IDE).
@TrystanLea what do you think?

TrystanLea, 2019-07-11:
I think this would be a nice feature, it would make sense to use an adaptation of the update scripts somehow to instigate install/update actions. Its not something we will get a chance to work on in the short term given other priorities but Im open to the idea and happy to receive a pull request if anyone wants to give it a go.

borpin, 2019-07-11:
The starting point is for the difference between core and non-core modules to be made clear and then for the modules with their own repos to consistently follow the master/stable model with a version number and a release tag.

Maintaining the release info could be done in the EmonScripts repo (which also needs a master/stable/version.txt structure.

chaveiro, 2021-10-02:
Solution here https://github.com/emoncms/emoncms/pull/1732

closed 2023-04-06 by TrystanLea