emoncms/emoncms #1167: Redis prefix in service-runner

borpin, 2019-01-20:
https://github.com/emoncms/emoncms/blob/e6da270246c7981210e668c81d448d9f052f5a20/scripts/services/service-runner/service-runner.py#L45

As discussed [here](https://community.openenergymonitor.org/t/development-roadmap-2019-robust-backup-process/9747/33), it was found that `service-runner` was not picking up the command from the backup module as `service-runner` was simply looking for `service-runner` and my settings had a redis prefix of `emoncms`.

When the python code was changed to look for `emoncms:service-runner` the backup module worked as expected.

borpin, 2019-01-20:
Just ran it, 6109!  Lots of..
```
   6088) "emoncms:input:5847"
   6089) "emoncms:input:lastvalue:603"
   6090) "emoncms:input:lastvalue:5208"
   6091) "emoncms:input:lastvalue:5218"
   6092) "emoncms:input:lastvalue:4874"
   6093) "emoncms:input:lastvalue:2602"
   6094) "emoncms:input:lastvalue:3647"
   6095) "emoncms:input:lastvalue:3396"
   6096) "emoncms:input:lastvalue:4586"
   6097) "emoncms:input:lastvalue:5706"
   6098) "emoncms:input:lastvalue:3133"
   6099) "emoncms:input:lastvalue:5687"
   6100) "emoncms:input:lastvalue:3473"
   6101) "emoncms:input:lastvalue:3862"
   6102) "emoncms:input:lastvalue:2005"
   6103) "emoncms:input:lastvalue:4949"
   6104) "emoncms:input:lastvalue:1432"
   6105) "emoncms:input:lastvalue:2230"
   6106) "emoncms:input:lastvalue:2540"
   6107) "emoncms:input:lastvalue:1847"
   6108) "emoncms:input:lastvalue:3713"
   6109) "emoncms:input:lastvalue:148"
```
so I wonder if something is not quite right.

borpin, 2019-01-20:
Could emoncms fire in a setup key that service-runner could scan for once (perhaps) and then service-runner will know what the prefix to use is?

greeebs, 2019-01-20:
> Could emoncms fire in a setup key that service-runner could scan for once (perhaps) and then service-runner will know what the prefix to use is
I think @pb66 had a better idea - to not use the prefix when sending stuff to service-runner.
@TrystanLea, @glynhudson what are your thoughts on this? Is there much work to remove the instance prefix when pushing values into redis with the `service-runner` name?

borpin, 2019-01-21:
> I think @pb66 had a better idea - to not use the prefix when sending stuff to service-runner.

I suspect that isn't easy.  Look at the code in the backup controller, it just pulls in `$redis` as a global object and uses it without specifying the prefix.  That must be done when it is instantiated.

https://github.com/emoncms/backup/blob/e8110759c46a62aaa86f70756c169465ac7acf16/backup/backup_controller.php#L38

greeebs, 2019-01-21:
> Could service-runner have its own prefix?

Or more ideally - not have any prefix. Then service-runner won't need changing ;)

borpin, 2019-01-22:
> Or more ideally - not have any prefix. Then service-runner won't need changing ;)

I think it was originally added as someone had an instance where they ran redis for a different purpose as well.

I also think that getting those Modules that need service-runner to create their own connection and thus use a predefined prefix might be the best solution.

In the short term, could an 'or' be added to the python code so it looks for both a key with and without the `emoncms` prefix as that is the default for non-pi installations?

greeebs, 2019-01-22:
> In the short term, could an 'or' be added to the python code so it looks for both a key with and without the emoncms prefix as that is the default for non-pi installations?

That's certainly a pragmatic interim solution. I'll put something together in the next few days and submit a PR.

TrystanLea, 2019-02-07:
That sounds like a good interim solution to me, happy to accept a pull request for that.

greeebs, 2019-02-07:
Apologies for not actually getting to that PR (and many thanks to @borpin for his!), too much paying work on just at the moment.

I did have another thought today about how we could make this more generic without bogging it down too much, but it does complicate the service a fair bit.
Given Brian's PR solves the immediate problem, but not the generic case, I think we leave this issue open and when I do get some time in the next few weeks, I'll submit a PR with more significant changes to get this working better in the "generic prefix", or multiple instance cases.

closed 2019-06-03 by borpin