emoncms/emoncms #1168: The redis default settings between emonpi and non emonpi are different.

borpin, 2019-01-20:
The redis default settings  for emonpi and non-emonpi installations are different.  This has caused an issue with the use of service runner.

https://github.com/emoncms/emoncms/blob/229365b46ccfe7fee75bbc5798ca6e6d09c8c9c6/default.settings.php#L19-L19

https://github.com/emoncms/emoncms/blob/229365b46ccfe7fee75bbc5798ca6e6d09c8c9c6/default.emonpi.settings.php#L18-L18

glynhudson, 2019-02-22:
Which setting would you recommend making consistent between both? Shall we standardize on "emoncms" as the prefix on the emonPi also?  It was assumed that on the emonPi there would be only one emoncms instance running

borpin, 2019-02-22:
Hi @glynhudson, the issue manifested itself in how service-runner was picking up the key (it didn't for self installs because of the prefix). I think removing is best on the assumption that someone who is savvy enough to be using REDIS for something else on the same machine will know they need to set it.

I think it does need a comment that service-runner only looks for no prefix or 'emoncms` as the prefix so if a prefix is required it is recommended to be 'emoncms'

TrystanLea, 2025-07-16:
This now defaults to emoncms on all installations I believe.