emoncms/emoncms #1180: Issues after upgrading to PHP 7.3

michaelpiron, 2019-01-25:
Dear all,

I upgraded my web server to PHP version 7.3. Since that moment, I noticed that my IotaWatt energy monitor acted differently when uploading data to EmonCMS running on my webserver.
I investigated a bit more in detail:
Apparently, upon initialization, the IotaWatt calls "https://[webserver URL]/emoncms/input/get?node=[IotaWatt node name]" 
--> in order to get the list of all defined inputs and the timestamp of the latest value posted.

Since upgrade to PHP 7.3, calling the above API link returns an error, stating there is an error in the preg_replace maskin the "emoncms/Modules/input/input_controller.php" code.
I'm not a PHP expert, but it seems like the masks in the different "preg_replace" function calls might have to be reviewed.

At this moment, I returned my webserver to PHP version 7.2, and the API calls act normally again, so something must have changed from version 7.2 to 7.3.

thib66, 2019-02-13:
Hello,
I have experienced some issues when upgrading to PHP 7.3 as well.
For instance in the My account section I was no longer able to change the language, the timezone; ...
I also returned my webserver to PHP version 7.2 in order to retrieve a normal behaviour of EmonCMS.
Regards,

thib66, 2019-02-13:
Hello Again,
in the Sources section a Node NoGroup has been created when installing PHP 7.3, containing a lot of sources with null ID 
[image]
Actually my web browser is no longer able to display the list of those numerous duplicated sources,
regards,

thib66, 2019-02-13:
[image]

thekazak, 2019-03-01:
Similar issue with a fresh new install on Debian testing (PHP 7.3) 

When trying to create/initialize device, I can see following errors in apache log file:
[Fri Mar 01 19:44:39.389687 2019] [php7:warn] [pid 28778] [client localhost:21725] PHP Warning:  preg_replace(): Compilation failed: invalid range in character class at offset 15 in /var/www/html/emoncms/Modules/device/device_model.php on line 334, referer: https://localhost:44311/emoncms/device/view
[Fri Mar 01 19:44:39.389704 2019] [php7:warn] [pid 28778] [client localhost:21725] PHP Warning:  preg_replace(): Compilation failed: invalid range in character class at offset 15 in /var/www/html/emoncms/Modules/device/device_model.php on line 340, referer: https://localhost:44311/emoncms/device/view
[Fri Mar 01 19:44:39.389719 2019] [php7:warn] [pid 28778] [client localhost:21725] PHP Warning:  preg_replace(): Compilation failed: invalid range in character class at offset 12 in /var/www/html/emoncms/Modules/device/device_model.php on line 346, referer: https://localhost:44311/emoncms/device/view
[Fri Mar 01 19:44:39.389736 2019] [php7:warn] [pid 28778] [client localhost:21725] PHP Warning:  preg_replace(): Compilation failed: invalid range in character class at offset 15 in /var/www/html/emoncms/Modules/device/device_model.php on line 131, referer: https://localhost:44311/emoncms/device/view

thekazak, 2019-03-09:
I found something, which seems to explains that issue with PHP 7.3 and PCRE2

https://bugs.php.net/bug.php?id=77334
> Perl treats a hyphen as a literal if it appears before or after a POSIX class
> (see below) or before or after a character type escape such as as \d or \H.
> However, unless the hyphen is the last character in the class, Perl outputs a
> warning in its warning mode, as this is most likely a user error. As PCRE2 has
> no facility for warning, an error is given in these cases.

Looks like regex should be rewritten in order to work with PHP 7.3

Escaping the hyphen makes it behave how expected it to behave.
echo preg_replace('/[^\d-\.]/', '', '123-abc'); -->
--> echo preg_replace('/[^\d\-\.]/', '', '123-abc');

thekazak, 2019-03-10:
Hi,
I tried escaping hyphen in the following files:
Modules/devices/device_model.php
Modules/devices/device_template.php
Modules/feed/feed_ model.php
Modules/input/input_ model.php

Ex1: 
@@ -433,7 +433,7 @@
         }

         if (isset($fields->nodeid)) {
-            if (preg_replace('/[^\p{N}\p{L}_\s-:]/u','',$fields->nodeid)!=$fields->nodeid) return array('success'=>false, 'message'=>'invalid characters in device nodeid');
+            if (preg_replace('/[^\p{N}\p{L}_\s\-:]/u','',$fields->nodeid)!=$fields->nodeid) return array('success'=>false, 'message'=>'invalid characters in device nodeid');
             $stmt = $this->mysqli->prepare("UPDATE device SET nodeid = ? WHERE id = ?");
             $stmt->bind_param("si",$fields->nodeid,$id);
             if ($stmt->execute()) {

Ex2:
@@ -78,8 +78,8 @@
     public function exists_nodeid_name($userid,$nodeid,$name)
     {
         $userid = (int) $userid;
-        $nodeid = preg_replace('/[^\p{N}\p{L}_\s-.]/u','',$nodeid);
-        $name = preg_replace('/[^\p{N}\p{L}_\s-.]/u','',$name);
+        $nodeid = preg_replace('/[^\p{N}\p{L}_\s\-.]/u','',$nodeid);
+        $name = preg_replace('/[^\p{N}\p{L}_\s\-.]/u','',$name);

         $stmt = $this->mysqli->prepare("SELECT id FROM input WHERE userid=? AND nodeid=? AND name=?");
         $stmt->bind_param("iss",$userid,$nodeid,$name);

I'm not quite sure if it needs escaping hyphens in regex for all files or just change some PHP setting to change behavior globally, but after the change above, at least creating new device now works

thekazak, 2019-03-10:
Hi,

here are the changes (escaping hyphen in regex), which I have done in order to have working inputs, feeds, devices created from external emonpi source
Distribution is debian unstable with PHP 7.3 
emoncms is latest from github

[emoncms-hyphen.diff.txt](https://github.com/emoncms/emoncms/files/2949273/emoncms-hyphen.diff.txt)

thekazak, 2019-03-10:
and here are all occurrences of hyphen in preg_replace used under Modules subtree
-:
-.
-?
replaced by
\-:
\-.
\-?

[emoncms-hyphen-ALL.diff.txt](https://github.com/emoncms/emoncms/files/2949345/emoncms-hyphen-ALL.diff.txt)

thib66, 2019-03-11:
Hello, a PR has been created in order to fix the PHP 7.3 issue with hyphen on regular expression. Based on changes proposed by @thekazak 
Please see #1221.
Regards,

glynhudson, 2019-04-30:
Thanks PR's have been merged 👍

closed 2019-04-30 by glynhudson