emoncms/emoncms #1198: duplicated inputs

pb66, 2019-02-08:
See https://community.openenergymonitor.org/t/double-input-entries-only-one-of-em-gets-data/1871?u=pb66

There is still a longstanding issue with inputs being duplicated, it's not reported quite so often these days, but I think that has to do with the MQTT input becoming more stable than the issue being resolved.

It isn't clear why or when it might occur, it isn't easily noticed when it does happen and it's not obvious to the average user how to resolve it .

pb66, 2019-03-03:
Also https://community.openenergymonitor.org/t/loss-of-inputs-following-upgrade/10290?u=pb66

pb66, 2019-05-05:
Also https://community.openenergymonitor.org/t/inputs-updates-the-feeds-do-not/10846?u=pb66

pb66, 2019-08-23:
Also https://community.openenergymonitor.org/t/duplicate-input-appears-spontaneously-result-lost-data/11736?u=pb66

borpin, 2019-08-23:
Got to be honest I have not seen this for a very long time and I pass a lot of MQTT data.

TrystanLea, 2020-01-21:
We fixed another cause of duplicate issues in this pull request https://github.com/emoncms/emoncms/pull/1356 I haven’t seen any occurrences since. Please re-open if you have seen examples of duplicate inputs since this was merged.

Bra1nK, 2021-01-17:
It's happened to me twice in the last few weeks following a reboot of the Pi

pb66, 2021-01-18:
reopened  for https://community.openenergymonitor.org/t/double-input-entries-only-one-of-em-gets-data/1871/8?u=pb66

borpin, 2021-01-19:
> It's happened to me twice in the last few weeks following a reboot of the Pi

@Bra1nK - can you be more specific about your setup -
* Hardware
* Emoncms version

If not an EmonPi or an Emonbase
* What is providing the Input data (emonhub for instance)
* Transport mechanism HTTP/MQTT

Bra1nK, 2021-01-19:
I'm on Emoncms Version 10.2.7 running on a Pi 4 booting from SSD and the affected input data is either from Node-Red using the node-red-node-emoncms node and simple json format or via the HTTP api (both are presumably using the same mechanism under the hood). The mechanism may not be significant as most of my inputs are via this route and only one or two were affected.

pb66, 2021-01-19:
Also issue #1651

Bra1nK, 2021-05-24:
Just an update to this. I'm rebooting on a daily basis now due to a daily backup strategy that seems to leave emoncms not working correctly after stopping services then restarting after the backup, the reboot after backup complete solves this issue but does frequently (not every time) result in duplicate inputs with the original not updating. Looking for a pragmatic solution I already have a node-red flow that checks for feeds not updated for > 1 hour and emails me so I've tagged on the 'clean inputs' api call to this flow so that all inputs without processes defined are deleted (http://[emoncms url]/input/clean). Works for me but may not suit everyone and I only lose an hour of data and don't have to continually look for duplicate inputs.

pb66, 2021-05-30:
Another report https://community.openenergymonitor.org/t/double-input-entries-only-one-of-em-gets-data/1871/11?u=pb66

TrystanLea, 2023-04-06:
For anyone still experiencing this, please feel free to reopen a new version of this issue with any recent occurrences, please provide as much detail as possible on how to replicate.  I'm not sure if recent changes have made any difference to this or not. I will certainly keep an eye out for it again.

closed 2023-04-06 by TrystanLea