emoncms/emoncms #1200: settings version notifications

pb66, 2019-02-11:
Both default settings files show a settings version of "10"
https://github.com/emoncms/emoncms/blob/9.9.5/default.emonpi.settings.php#L175-L175
https://github.com/emoncms/emoncms/blob/9.9.5/emonpi.settings.php#L174-L174

But the process_settings only generates a warning if the current version is "<9"
https://github.com/emoncms/emoncms/blob/9.9.5/process_settings.php#L66-L66

Is this intentional? ie are versions 9 and 10 essentially the same and interchangeable? 

Or were the settings templates nudged along without the process_settings.php file being updated in error?

TrystanLea, 2019-02-12:
Looking at the commits, the first here for default.settings.php does not give much info:
https://github.com/emoncms/emoncms/blame/b21549e35d6bc057e59db4f11957ed47b3da0731/default.settings.php#L178

But there is a discussion here associated with the process_settings.php commit, where Glyn talks about the emonpi update method and a md5 comparison:
https://github.com/emoncms/emoncms/pull/648

There was an addition of a new settings:

     $allow_config_env_vars = false; 

but process_settings still checked for its existence and so was not a breaking change requiring the user to update their settings file.

The change from version 7 to 8 looks like a more significant change: https://github.com/emoncms/emoncms/commit/818905ba0bb681a724e285d486cedeab4c9f47fa that perhaps needed user intervention..

It would make sense to update the config version number on each change to the default settings but only update the process_settings version requirement where user intervention is required. @chaveiro @glynhudson is that what is intended here?

chaveiro, 2019-02-12:
Hi the intended functionality for the config version variable is to alert an upgrading user for upgrading the config file when new variables or breaking changes appear on the upgraded version.
When the developer performes such change the version on the default config  must be increased and reflected on process_settings.php so that the alert  works as expected.

borpin, 2019-02-13:
Of course, if we introduced hierarchical settings, there would be less of an issue...

I'll get my coat.... [GRD]

TrystanLea, 2019-07-11:
With hierarchical settings on its way, I will close this issue.

closed 2019-07-11 by TrystanLea