emoncms/emoncms #1202: Mariadb will not start after a dist-upgrade

borpin, 2019-02-13:
Discussion https://community.openenergymonitor.org/t/mariadb-cannot-connect-after-dist-upgrade/10071/4

Solution;

Change in file /lib/systemd/system/mariadb.service the value:
```
ProtectHome=true
```
to
```
ProtectHome=false
```

It may be that the dist-upgrade has introduced a new service file that now has the value as **true** or else this file is edited in the EmonSD build (which is still a closely guarded secret).

Editing **any** service file is a **bad idea**. Instead;
```
sudo systemctl edit mariadb.service
```
then add
```
[Service]
ProtectHome=false
```
Save file then;
```
sudo systemctl daemon-reload
sudo systemctl restart mariadb.service
```
This change is then protected if the service file is updated.

borpin, 2019-02-15:
2 users have now reported this issue.

alexandrecuer, 2019-02-15:
I also have the issue
the solution I found is the following :
```
sudo nano /lib/systemd/system/mariadb.service
```
Change``ProtectHome=true`` to ``ProtectHome=false``

Then reload
```
sudo systemctl daemon-reload
sudo service mariadb start
sudo shutdown -r now
```

borpin, 2019-02-15:
@alexandrecuer - The problem with fixing it by editing the service file directly, is that if the package maintainer decides to update the service file, your changes will be overwritten.  If you use my method above, they will not, so it is a permanent fix.

It is regarded as a bad practice to edit any file in `/lib/systemd` unless you created it.

alexandrecuer, 2019-02-15:
@borpin - Ok seen, you're right, your solution is the good one...I did not think of that

borpin, 2019-02-16:
On checking the service file pre upgrade has the setting to false.  The question is, as part of the SDCard Build, it the service file edited?

I am pretty certain it has as it has a significantly more recent datestamp.

Moral of story **Don't Edit Service Files**

TrystanLea, 2019-03-04:
Discussing with Glyn, we are not aware of having changed the mariadb.service file on emonSD. Do we need to issue an update with a version that can be upgraded?

borpin, 2019-03-04:
It would appear then that this is a change on a dist-upgrade by the package to change this setting to true by default (security probably).  You could check by checking the service file before doing a dist-upgrade and then after and see if it has been changed (not had a chance to).

If this is correct I would be inclined to leave it, as automating a change may break existing setups and fix installations as they arise.

Might be worth rebuilding a new SD image with a dist-upgrade done and this fix in place.

dist-upgrade is the recommended method to upgrade now in the official docs.

pb66, 2019-05-02:
> Discussing with Glyn, we are not aware of having changed the mariadb.service file on emonSD

Does this ring any bells? https://github.com/emoncms/emoncms/commit/428e7244785a35ee93365f42be8f1e4e021ee8cd

https://github.com/emoncms/emoncms/blob/5e21b308b67f9ce1be95b7c78c484b0ce693c6a8/docs/RaspberryPi/read-only.md#move-mysql-database.

This highlights the potential (or maybe now realised) issue with customising service units that are managed by a package manager (even when done apparently "correctly"). This change must have been in place before the apt-get upgrade or mysql would not have started, so it must of been reverted by the upgrade.

glynhudson, 2019-05-02:
>It may be that the dist-upgrade has introduced a new service file that now has the value as true or else this file is edited in the EmonSD build (which is still a closely guarded secret).

Yes, `ProtectHome=true` was set during the emonSD card build as documented here https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/read-only.md

@borpin please don't spread misinformation that the emonSD card build as a 'closely guarded secret'. It certainly is not. We have documenting all the steps taken to build the image. It's true the docs could be better, we're working on improving this. @TrystanLea has recently put a lot of work into publishing build scripts for the image. 

Is it possibly to apply this change in a way which will not be changed automatically by package upgrades in the future? 

I've just `apt-get update && apt-get upgrade` and the file `sudo apt-get update && sudo apt-get upgrade` has not changed and mariadb is still working. 

After upgrade, I'm running:

`mariadb  Ver 15.1 Distrib 10.1.38-MariaDB, for debian-linux-gnueabihf (armv7l) using readline 5.2`

It must be `dist-upgrade` which causes the issue. What's the advantage of running `dist-upgrade`? I usually avoid it for this reason since `upgrade` applies all the important security updates required. 

From what I've read `dist-upgrade` is not recommended for important systems unless specifically required. From [digitalocean](https://www.digitalocean.com/community/questions/general-best-practices-for-apt-get-upgrade-and-dist-upgrade):

>SAFE: The "apt-get upgrade" command will normally only install updates (or fixes) to currently installed packages. This is the safest command for keeping the server's software, security patches, and minor changes current. The apt-get upgrade process will NOT generally install new releases, where major changes (including removal of packages or GRUB update is required). For example, when a new Linux kernel (linux-image-3.x.x-xx-generic, etc.) is available, the package will not get installed. A number of cloud hosting providers upgrade their servers automatically using this basic "upgrade" process.

>SOME RISKS: The "apt-get dist-upgrade" command is for upgrading from one stable Debian/Ubuntu distribution to the next. It also upgrades packages and settings in a more complex and comprehensive way, including deleting, or deleting and replacing, some packages. In order to install an updated "new" kernel, or new release, you will need to run this command. You are occasionally notified of the opportunity to do this when you run apt-get upgrade and it reports that certain packages have been "keep back". HOWEVER: (1) Ubuntu officially recommends running "do-release-upgrade" instead of "dist-upgrade" because the former has the ability to handle system configuration changes sometimes needed between releases; and (2) Doing dist-upgrade/do-release-upgrade may not be supported by your cloud hosting company because of hypervisor/cloud-vps-instance configuration and management restrictions. Serious situations can arise, regarding Grub for example, and you need to make certain you have console access in case something happens. If you perform these types of advanced software/kernel-version upgrades, you may find that your hosting company will not provide support, leaving you in isolation with no way to go back, and an uneasy path forward. A dist-upgrade situation recently plagued certain Debian/Ubuntu cloud instances because of a Grub update question popping up during the upgrade process which could not be confidently answer. In response, some providers said that they DID NOT support dist-upgrades of their servers !

>BEST PRACTICE: "Best Practice" is NOT to do either a "dist-upgrade" or "dist-upgrade". This level of upgrade is seldom if ever needed, and not at all required to keep your system up to date, secure, and running smoothy. If you must move to a later release, follow the advice of "prezzy_dent" below, and do it through the DigitalOcean settings (if possible), or upon their advice only.

borpin, 2019-05-03:
> We have documenting all the steps taken to build the image.

I'm not aware of any documentation re the Oct18 image.  If you can point me to it, it would help.

`dist-upgrade` is the officially recommended means of upgrading Raspbian https://www.raspberrypi.org/documentation/raspbian/updating.md

> From what I've read dist-upgrade is not recommended for important systems unless specifically required. From digitalocean:

That item was written in 2014.

> This highlights the potential (or maybe now realised) issue with customising service units that are managed by a package manager (even when done apparently "correctly").

> Yes, ProtectHome=true was set during the emonSD card build

Because the original method (and that employed by the SDCard image) was to actually change the service file rather than use a drop in, **any** upgrade of mariadb could change that setting.

This demonstrates the need to **not** actually edit system files but to use the extremely good, builtin mechanisms to use drop-in files to override settings.  A simlilar issue exists with LogRotate.

All this discussion does not solve the overriding problem.  Emoncms is broken by a `dist-upgrade` and it is not just the `ProtectHome=true` setting.

glynhudson, 2019-05-03:
>I'm not aware of any documentation re the Oct18 image. If you can point me to it, it would help.

It was build using the same documented setup just without the read/only partition. See Trystan's build scripts: https://github.com/openenergymonitor/emonpi/tree/master/install

>dist-upgrade is the officially recommended means of upgrading Raspbian https://www.raspberrypi.org/documentation/raspbian/updating.md

I would assume this is because the foundation want to ensure users have the latest software, this makes sense for education etc especially when using their desktop applications. However, dist-upgrade is not recommended for critical systems. At least not without careful backups beforehand. I can't see any advantage for emon users to run dist-update since upgrade applies any security updates to installed packages which is obviously very important. 

Reliability is the most important thing for logging applications, emonSD should prioritise security and stability. I don't see the need to burden the users with the unnecessary risk of dist-upgrade. Even without editing service files there is a high probably that at some point dist-upgrade will break something, I think it's best to recommend `apt-get upgrade` to apply security updates and update install packages. For the same reason `raspi-upgrade` is not recommended, RasPi firmware updates will eventually be packaged into the package manager when stable. From [raspi-update readme:](https://github.com/Hexxeh/rpi-update)

>Even on Raspbian you should only use this with a good reason. This gets you the latest bleeding edge kernel/firmware. There is always the possibility of regressions. Bug fixes and improvements will eventually make their way into new Raspbian releases and apt-get when they are considered sufficiently well tested.

Agree, it would be best to avoid editing service files if possible. Do you have a suggestion how this `ProtectHome=false` modification can be achieved by not editing service files?

>All this discussion does not solve the overriding problem. Emoncms is broken by a dist-upgrade and it is not just the ProtectHome=true setting.

What else get's broken? I have not heard reports of anything else being broken by `dist-upgrade`

pb66, 2019-05-03:
This is still as clear as mud!

> Yes, ProtectHome=true was set during the emonSD card build as documented here https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/read-only.md

For mysql to work from the /home/pi/data folder the setting needs to be `ProtectHome=false` not `ProtectHome=true` is the above comment a typo @glynhudson? The link you provide not only demonstrates the setting being set to `ProtectHome=false` but it is a ducument for "read-only" which the Oct 2018 image is not! You can see where confusion seeps in.

>>This highlights the potential (or maybe now realised) issue with customising service units that are managed by a package manager (even when done apparently "correctly").
> Because the original method (and that employed by the SDCard image) was to actually change the service file rather than use a drop in, any upgrade of mariadb could change that setting.
>
>This demonstrates the need to not actually edit system files but to use the extremely good, builtin mechanisms to use drop-in files to override settings.

I suspect you have not properly read or digested what I said @borpin! The guide shows the correct systemctl edit function being used, hence why I said "(even when done apparently "correctly")" as this raises questions over whether the guide was adhered to and if it was, then editing service files "**even correctly**" would still appear to fall foul of an upgrade, **_IF_** it has been done as documented.

I agree the dist-upgrade is the correct method of upgrading. I would always recommend that as the preferred method, unless there is very good (and known!) reason to do otherwise.

If the dist-upgrade should not be used with the emonSD, that should be widely documented, perhaps in the warnings when a console is opened! Or perhaps even blocked. The bottom line is that systems cannot be left un-updated, so by not including an apt-get update in the emonSD update process, users are being forced to make their own decision, this of course also conflicts with the emonSD position of "not needing to use SSH" as that creates potential security concerns.

[edit]

Ok so I decided to check this myself. On the emonSD image I have been looking at L2R with, it has not been updated via apt-get at anytime, so I looked at the maria.db service file and it is set to `ProtectHome=false` directly in the original unit file! So I edited that line back to original and tested mariadb, it failed to open files in /home/pi as expected (all appropriate reload-daemon's and restarts done). I then added the edited setting via the documented method of using "systemctl edt". This returned things to a "fixed" state (as expected). I then did a dist-upgrade and rechecked the mariadb service, it appears to be working fine with no errors (as expected?) so it looks like this is a case of shooting one's self in the foot, the documented method has not been adhered to, if it had, this fault would not of occurred.

If there's any interest, the actual changes (diff to copy of original maria.db service as it was intended, ie `ProtectHome=true`) made to the original mariadb service unit by the package are

```
pi@emonpi:~ $ diff /home/pi/mariadb.service /lib/systemd/system/mariadb.service
16c16,18
< Description=MariaDB database server
---
> Description=MariaDB 10.1.38 database server
> Documentation=man:mysqld(8)
> Documentation=https://mariadb.com/kb/en/library/systemd/
78c80
< # Use the [service] section and Environment="MYSQLD_OPTS=...".
---
> # Use the [Service] section and Environment="MYSQLD_OPTS=...".
87d88
< KillMode=process
105c106,107
< ## and adding/setting the following will override this file's settings.
---
> ## and adding/setting the following under [Service] will override this file's
> ## settings.
```
Obviously, had I not edited the service unit back to `ProtectHome=true` prior to upgrading, the diff would also show the `ProtectHome=false` reverted to `ProtectHome=true` as well.

[edit2] I started writing this post much earlier in the day, but have had to keep popping out so some of the above overlaps you comments both here and on the forum Glyn. The bottom line is that if the service unit had been edited correctly using "systemctl edit" as documented on the page both you and I have linked. This would not be an issue, even with dist-upgrade.

glynhudson, 2019-05-03:
A solution has been suggested by @pb66 on the forum:

```
sudo systemctl edit mariadb.service
```
then to that blank file add
```
[Service]
ProtectHome=false
```
and save with ctrl-x, y, enter.

Then reload the service and restart
```
sudo systemctl daemon-reload
sudo systemctl restart mariadb
```

https://community.openenergymonitor.org/t/warning-do-not-do-apt-get-dist-upgrade/10823/18?u=glyn.hudson

I will test this and look at implementing a fix for emonSD users

glynhudson, 2019-05-03:
Fixed by https://github.com/openenergymonitor/emonpi/commit/59faeea2494fd27f76b66af04d73983917c3488c

closed 2019-05-03 by glynhudson

borpin, 2019-05-05:
@pb66 
> The guide shows the correct systemctl edit function being used

Check the history for when that instruction was created 😄

borpin, 2019-05-05:
@glynhudson 
> I would assume this is because the foundation want to ensure users have the latest software, this makes sense for education etc especially when using their desktop applications. However, dist-upgrade is not recommended for critical systems.

Running anything on an RPi is not a 'critical system'.  The RPi may be doing a number of different things (the EmonPi even has Node-Red preloaded) which **need** a dist-upgrade.  Anyone who then thinks 'How do I upgrade Raspbian' is going to come across the **Official** suggestion of a `dist-upgrade` (and BTW, the option of *just* doing an `upgrade` is not mentioned).  As the current emoncms update process makes no effort to update the underlying OS, the possibility of a user doing a dist-upgrade is high.  I suggest emoncms should be tested so it continues to work in these circumstances.

pb66, 2019-05-05:
> Check the history for when that instruction was created

? I did, and posted the times 2 days ago on the forum as you know

> [This was originally submitted as a PR by @bwduncan on 29th Dec 2018 and merged by you that same day.](https://community.openenergymonitor.org/t/warning-do-not-do-apt-get-dist-upgrade/10823/21)

And further explained earlier this morning

> [It’s only due to diving into the commit history that I discovered these changes were made after the release of the Oct 2018 image.](https://community.openenergymonitor.org/t/warning-do-not-do-apt-get-dist-upgrade/10823/27?u=pb66)

Further to your comments above, if we were to use upgrade in place of dist-upgrade it would limit the operational life of an emonPi (image) to whatever the LTS is for security updates and we might all have still been running wheezy!  Even stretch LTS expires in 2022, if you do not use dist-upgrade, it will remain "stretch" for ever, even when support expires.

pb66, 2019-05-07:
PR submitted for revised fix #104

glynhudson, 2019-05-07:
>Running anything on an RPi is not a 'critical system'. The RPi may be doing a number of different things (the EmonPi even has Node-Red preloaded)

Degrees of criticality can be debated, but personally I hate to lose even 1hr of logging data since it messes up the energy totals for that period. NodeRED is not pre-installed anymore. 

>the possibility of a user doing a dist-upgrade is high. I suggest emoncms should be tested so it continues to work in these circumstances.

Sure, agree it's a good idea for us to test running `dist-upgrade` 

> if we were to use upgrade in place of dist-upgrade it would limit the operational life of an emonPi (image) to whatever the LTS is for security updates and we might all have still been running [...]

I would recommend that in 2022 the user would be best to back up their data and move to a fresh SD card with the latest emonSD card build with a vanilla build of Debian Buster. 

Each user is free to upgrade their system as they see fit, however personally I will recommend user run `upgrade` to apply security updates unless there is a particular reason why `dist-upgrade` is required. I think it's unnecessary risky for the average users to run `dist-upgrade` on a whim. 

I would like to explore applying OS security updates as part of the emonPi update process e.g https://www.3cx.com/blog/voip-howto/os-security-updates-debian/. What do you think to this? e.g

```
grep security /etc/apt/sources.list | tee /etc/apt/security.sources.list
apt-get update
apt-get upgrade -o Dir::Etc::SourceList=/etc/apt/security.sources.list
```